ov2slam/ov2slam #4: Could not link library libov2slam

neo-knight-td, 2021-03-03:
Hi all,
I'm trying to get ov2slam running on my Ubuntu 20.04 machine. I work with a version of Ceres already installed. When I'm running command  `catkin_make --pkg ov2slam`, I get the following error : 

`-- Configuring done
CMake Warning (dev) at ov2slam/CMakeLists.txt:127 (add_library):
  Policy CMP0028 is not set: Double colon in target name means ALIAS or
  IMPORTED target.  Run "cmake --help-policy CMP0028" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  Target "ov2slam" links to target "Ceres::ceres" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /home/thomas/catkin_ws/build
####
#### Running command: "make -j8 -l8" in "/home/thomas/catkin_ws/build/ov2slam"
####
[ 14%] Building CXX object ov2slam/CMakeFiles/ov2slam.dir/src/camera_visualizer.cpp.o
[ 14%] Building CXX object ov2slam/CMakeFiles/ov2slam.dir/src/visual_front_end.cpp.o
[ 14%] Building CXX object ov2slam/CMakeFiles/ov2slam.dir/src/ov2slam.cpp.o
[ 28%] Building CXX object ov2slam/CMakeFiles/ov2slam.dir/src/frame.cpp.o
[ 28%] Building CXX object ov2slam/CMakeFiles/ov2slam.dir/src/slam_params.cpp.o
[ 28%] Building CXX object ov2slam/CMakeFiles/ov2slam.dir/src/camera_calibration.cpp.o
[ 42%] Building CXX object ov2slam/CMakeFiles/ov2slam.dir/src/feature_tracker.cpp.o
[ 42%] Building CXX object ov2slam/CMakeFiles/ov2slam.dir/src/feature_extractor.cpp.o
[ 42%] Building CXX object ov2slam/CMakeFiles/ov2slam.dir/src/map_manager.cpp.o
[ 57%] Building CXX object ov2slam/CMakeFiles/ov2slam.dir/src/map_point.cpp.o
[ 57%] Building CXX object ov2slam/CMakeFiles/ov2slam.dir/src/mapper.cpp.o
[ 57%] Building CXX object ov2slam/CMakeFiles/ov2slam.dir/src/multi_view_geometry.cpp.o
[ 71%] Building CXX object ov2slam/CMakeFiles/ov2slam.dir/src/ceres_parametrization.cpp.o
[ 71%] Building CXX object ov2slam/CMakeFiles/ov2slam.dir/src/optimizer.cpp.o
[ 71%] Building CXX object ov2slam/CMakeFiles/ov2slam.dir/src/estimator.cpp.o
[ 85%] Building CXX object ov2slam/CMakeFiles/ov2slam.dir/src/loop_closer.cpp.o
[ 85%] Linking CXX shared library /home/thomas/catkin_ws/devel/lib/libov2slam.so
/usr/bin/ld: cannot find -lCeres::ceres
collect2: error: ld returned 1 exit status
make[2]: *** [ov2slam/CMakeFiles/ov2slam.dir/build.make:524: /home/thomas/catkin_ws/devel/lib/libov2slam.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:3761: ov2slam/CMakeFiles/ov2slam.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j8 -l8" failed`

I tried reinstalling Ceres, use the third-party version of Ceres but I always get the same error. Seems like code is built but link to library libov2slam can't be made (I checked and this lib does not appear on the designated location on my PC ( /home/thomas/catkin_ws/devel/lib/ ).

Anyone encountered this already ? Or does someone have an idea on how to solve the problem ?

Thank you !

ferreram, 2021-03-04:
Hi @neo-knight-td,

Have you used the build_thirdparty.sh file to build the thirdparty version of Ceres?
If so, could you try running this command : "catkin_make --force-cmake --pkg ov2slam".

neo-knight-td, 2021-03-04:
Hi @ferreram,

Thank you for your reply. Unfortunately, I've tried your command but still getting the same error : 

`CMake Warning (dev) at ov2slam/CMakeLists.txt:127 (add_library):
  Policy CMP0028 is not set: Double colon in target name means ALIAS or
  IMPORTED target.  Run "cmake --help-policy CMP0028" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  Target "ov2slam" links to target "Ceres::ceres" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /home/thomas/catkin_ws/build
####
#### Running command: "make -j8 -l8" in "/home/thomas/catkin_ws/build/ov2slam"
####
[  0%] Linking CXX shared library /home/thomas/catkin_ws/devel/lib/libov2slam.so
/usr/bin/ld: cannot find -lCeres::ceres
collect2: error: ld returned 1 exit status
make[2]: *** [ov2slam/CMakeFiles/ov2slam.dir/build.make:524: /home/thomas/catkin_ws/devel/lib/libov2slam.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:3761: ov2slam/CMakeFiles/ov2slam.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j8 -l8" failed`

ferreram, 2021-03-04:
When running catkin_make with --force-cmake, you should see a message if Ceres is found :
```
if (Ceres_FOUND)
  message(STATUS "Found Ceres lib : ${CERES_VERSION} - ${Ceres_DIR}")
endif (Ceres_FOUND)
```
If this message does not appear, it means that cmake did not find the ceres package.  As Ceres is first searched for here:
`find_package(Ceres NO_DEFAULT_PATH PATHS ${PROJECT_SOURCE_DIR}/Thirdparty/ceres-solver/install/)`

Could you check if this install directory exists and if it contains an "include" and "lib" directories?

neo-knight-td, 2021-03-04:
I see that Ceres is found : 

`-- +++ processing catkin package: 'ov2slam'
-- ==> add_subdirectory(ov2slam)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- iBoW-LCD found!  Going to use Loop Closer!
-- Found Eigen3 lib : 3.3.7 - /usr/include/eigen3
CMake Warning (dev) at ov2slam/CMakeLists.txt:96 (get_target_property):
  Policy CMP0045 is not set: Error on non-existent target in
  get_target_property.  Run "cmake --help-policy CMP0045" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  get_target_property() called with non-existent target "opengv".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found installed version of Eigen: /usr/lib/cmake/eigen3
-- Found required Ceres dependency: Eigen version 3.3.7 in /usr/include/eigen3
-- Found required Ceres dependency: glog
-- Found installed version of gflags: /usr/lib/x86_64-linux-gnu/cmake/gflags
-- Detected gflags version: 2.2.2
-- Found required Ceres dependency: gflags
-- Found Ceres version: 1.14.0 installed in: /usr/local with components: [EigenSparse, SparseLinearAlgebraLibrary, LAPACK, SuiteSparse, CXSparse, SchurSpecializations, OpenMP, Multithreading]
-- Found installed version of Eigen: /usr/lib/cmake/eigen3
-- Found required Ceres dependency: Eigen version 3.3.7 in /usr/include/eigen3
-- Found required Ceres dependency: glog
-- Found installed version of gflags: /usr/lib/x86_64-linux-gnu/cmake/gflags
-- Detected gflags version: 2.2.2
-- Found required Ceres dependency: gflags
-- Found Ceres version: 1.14.0 installed in: /usr/local with components: [EigenSparse, SparseLinearAlgebraLibrary, LAPACK, SuiteSparse, CXSparse, SchurSpecializations, OpenMP, Multithreading]
-- Found Ceres lib : 1.14.0 - /usr/local/lib/cmake/Ceres
-- Found Sophus lib : 1.1.0 - /home/thomas/catkin_ws/src/ov2slam/Thirdparty/Sophus/build`

Although it found my version of Ceres and not the one installed by the Thirdparty builder. Regarding  ```/Thirdparty/ceres-solver/install/```, this directory does contain the include and lib dirs.

ferreram, 2021-03-04:
Ok.  Could you go to the build directory of your workspace and erase the "ov2slam" directory?  This way, when you will run again "catkin_make --pkg ov2slam", cmake should find the ceres package located within the Thirdparty repository.

neo-knight-td, 2021-03-04:
It did not find the Third-party version of Ceres : 

`-- Configuring done
CMake Warning (dev) at ov2slam/CMakeLists.txt:127 (add_library):
  Policy CMP0028 is not set: Double colon in target name means ALIAS or
  IMPORTED target.  Run "cmake --help-policy CMP0028" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  Target "ov2slam" links to target "Ceres::ceres" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /home/thomas/catkin_ws/build
####
#### Running command: "make -j8 -l8" in "/home/thomas/catkin_ws/build/ov2slam"
####
Scanning dependencies of target ov2slam
[ 14%] Building CXX object ov2slam/CMakeFiles/ov2slam.dir/src/ov2slam.cpp.o
[ 14%] Building CXX object ov2slam/CMakeFiles/ov2slam.dir/src/camera_visualizer.cpp.o
[ 14%] Building CXX object ov2slam/CMakeFiles/ov2slam.dir/src/slam_params.cpp.o
[ 28%] Building CXX object ov2slam/CMakeFiles/ov2slam.dir/src/frame.cpp.o
[ 14%] Building CXX object ov2slam/CMakeFiles/ov2slam.dir/src/visual_front_end.cpp.o
[ 28%] Building CXX object ov2slam/CMakeFiles/ov2slam.dir/src/camera_calibration.cpp.o
[ 42%] Building CXX object ov2slam/CMakeFiles/ov2slam.dir/src/feature_extractor.cpp.o
[ 42%] Building CXX object ov2slam/CMakeFiles/ov2slam.dir/src/feature_tracker.cpp.o
[ 42%] Building CXX object ov2slam/CMakeFiles/ov2slam.dir/src/map_manager.cpp.o
[ 57%] Building CXX object ov2slam/CMakeFiles/ov2slam.dir/src/map_point.cpp.o
[ 57%] Building CXX object ov2slam/CMakeFiles/ov2slam.dir/src/mapper.cpp.o
[ 57%] Building CXX object ov2slam/CMakeFiles/ov2slam.dir/src/multi_view_geometry.cpp.o
[ 71%] Building CXX object ov2slam/CMakeFiles/ov2slam.dir/src/ceres_parametrization.cpp.o
[ 71%] Building CXX object ov2slam/CMakeFiles/ov2slam.dir/src/optimizer.cpp.o
[ 71%] Building CXX object ov2slam/CMakeFiles/ov2slam.dir/src/estimator.cpp.o
[ 85%] Building CXX object ov2slam/CMakeFiles/ov2slam.dir/src/loop_closer.cpp.o
[ 85%] Linking CXX shared library /home/thomas/catkin_ws/devel/lib/libov2slam.so
/usr/bin/ld: cannot find -lCeres::ceres
collect2: error: ld returned 1 exit status
make[2]: *** [ov2slam/CMakeFiles/ov2slam.dir/build.make:524: /home/thomas/catkin_ws/devel/lib/libov2slam.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:3761: ov2slam/CMakeFiles/ov2slam.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j8 -l8" failed`

ferreram, 2021-03-04:
Could you add the --force-cmake flag to see the path of the Ceres package that was found by cmake?

neo-knight-td, 2021-03-04:
Path found by cmake did not change. 
`-- +++ processing catkin package: 'ov2slam'
-- ==> add_subdirectory(ov2slam)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- iBoW-LCD found!  Going to use Loop Closer!
-- Found Eigen3 lib : 3.3.7 - /usr/include/eigen3
CMake Warning (dev) at ov2slam/CMakeLists.txt:96 (get_target_property):
  Policy CMP0045 is not set: Error on non-existent target in
  get_target_property.  Run "cmake --help-policy CMP0045" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  get_target_property() called with non-existent target "opengv".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found installed version of Eigen: /usr/lib/cmake/eigen3
-- Found required Ceres dependency: Eigen version 3.3.7 in /usr/include/eigen3
-- Found required Ceres dependency: glog
-- Found installed version of gflags: /usr/lib/x86_64-linux-gnu/cmake/gflags
-- Detected gflags version: 2.2.2
-- Found required Ceres dependency: gflags
-- Found Ceres version: 1.14.0 installed in: /usr/local with components: [EigenSparse, SparseLinearAlgebraLibrary, LAPACK, SuiteSparse, CXSparse, SchurSpecializations, OpenMP, Multithreading]
-- Found installed version of Eigen: /usr/lib/cmake/eigen3
-- Found required Ceres dependency: Eigen version 3.3.7 in /usr/include/eigen3
-- Found required Ceres dependency: glog
-- Found installed version of gflags: /usr/lib/x86_64-linux-gnu/cmake/gflags
-- Detected gflags version: 2.2.2
-- Found required Ceres dependency: gflags
-- Found Ceres version: 1.14.0 installed in: /usr/local with components: [EigenSparse, SparseLinearAlgebraLibrary, LAPACK, SuiteSparse, CXSparse, SchurSpecializations, OpenMP, Multithreading]
-- Found Ceres lib : 1.14.0 - /usr/local/lib/cmake/Ceres
-- Found Sophus lib : 1.1.0 - /home/thomas/catkin_ws/src/ov2slam/Thirdparty/Sophus/build
`

ferreram, 2021-03-04:
Hum that's weird.
Could you run the following commands:
```
cd ~/catkin_ws/
rm -r build/ov2slam
rm -r devel/ov2slam
catkin_make --force-cmake --pkg ov2slam
```

neo-knight-td, 2021-03-04:
I have no directory ov2slam in my devel folder. And I am still getting the same error :/

ferreram, 2021-03-04:
Yes I was expecting that the ov2slam folder in the devel directory did not exist, it was to be sure.

Could you comment this line in the CMakelists : 
``` find_package(Ceres REQUIRED)  ```

neo-knight-td, 2021-03-04:
Still same error. I think this is related to that part of the CMakelists.txt : 

`target_link_libraries(${PROJECT_NAME}
    PUBLIC
      ${catkin_LIBRARIES}
      Eigen3::Eigen
      Sophus::Sophus
    PRIVATE
      ${OpenCV_LIBS}
      Ceres::ceres
)`

ferreram, 2021-03-04:
No but I don't understand why cmake is always finding you own ceres package in "usr/local/lib".  Is it still the case?  I mean, now that you commented out the line `find_package(Ceres REQUIRED) `, if you erase the ov2slam folder from "~/catkin_ws/build/" and then run "catkin_make", is the found ceres package still located in "usr/local/lib" ?

Otherwise, you can try changing "Ceres::ceres" for "${CERES_LIBRARIES}".

neo-knight-td, 2021-03-04:
Great !

Changing "Ceres:ceres" to "${CERES_LIBRARIES}" line 162 of CMakeLists.txt solved the problem ! No more error nor warning ;) 

Thank you @ferreram !

closed 2021-03-04 by neo-knight-td