ov2slam/ov2slam #5: Sudden scale drift on top down forward motion

laxnpander, 2021-03-05:
Hey hey,

first of all ov2slam looks like a pretty promising SLAM for my application. The quick initialization and performance in the first few seconds is very nice. However, I experience very sudden scale drifts out of the blue. I made a quick video and attached a link to this in the bottom.
My question is: What can I do about it? I already chose the high accuracy settings, but it seems like something gets out of hand really fast.  My camera setup is rather uncommon for the average SLAM dataset with rather low frame rate (7 Hz) and high resolution (1920x1080) with images looking top down from a forward flying drone. Any suggestions? I'd prefer robustness over fast performance.

https://drive.google.com/file/d/1k9GIVHZNiTi8cD-KGqulIvPmsWlW1TrU/view?usp=sharing

ferreram, 2021-03-05:
Hi,
My first guess would be to increase the KLT pyramid number of levels to 4 or 5 and to increase the size of the tracking window, maybe up to 21x21 px.  Also, with such a low frame rate, the constant velocity motion model used in the front-end is likely to be wrong, so enforcing the use of P3P for pose estimation might be a good shot.

I might get a better grasp of what's happening when I see your video.  I have sent you a request access.

ferreram, 2021-03-05:
I have watched your video.  It's hard to tell anything from the trajectory only, the image_track topic could help more seeing if anything goes wrong in the tracking.

ferreram, 2021-03-05:
Also, I noticed that the reconstructed 3D map seems to be very planar (I guess the video sequence is recorded from a drone flying at high altitude).  In such scenarios, the computation of the Essential Matrix might get in a degenerate case.  Replacing the 2D-2D filtering stage with a RANSAC based on a Homography might better work here.

laxnpander, 2021-03-05:
@ferreram Ah sorry, I meant it to be available straightaway. I did what you suggested and also tuned all parameters that looked accuracy related. It massively improved the overall results. I am quite happy now actually.

I have datasets on different altitudes, this is one at 150m:
https://drive.google.com/file/d/1Vcf6q5NhY-4Mmy-cc6m6dx58B4BHTNO4/view?usp=sharing

This is one at 30m:
https://drive.google.com/file/d/1owq0En-thNqPitIYq7sANDkcwhH-Ydun/view?usp=sharing

The 150m one was the same as my initial video. Didn't want to bore you, so it's not the full run, but it does not drift over the whole run. The second one has more problems, but to be honest it's a very challenging dataset. I don't expect it to run perfectly on that. Your SLAM seems to be outperforming ORB SLAM2 massively on this kind of data. Very good job!

closed 2021-03-05 by laxnpander