ov2slam/ov2slam #6: Cannot catkin_make undefined reference libov2slam.so

neo-knight-td, 2021-03-08:
Hi all,

I'm trying to build ov2slam on a Jetson TX2 within ROS melodic. I got the following error when I try the command ```catkin_make --pkg ov2slam``` : 

`/usr/bin/ld: warning: libopencv_core.so.3.4, needed by /home/thomas/catkin_ws/devel/lib/libov2slam.so, may conflict with libopencv_core.so.3.2
/usr/bin/ld: warning: libopencv_imgproc.so.3.4, needed by /usr/local/lib/libopencv_calib3d.so.3.4, may conflict with libopencv_imgproc.so.3.2
CMakeFiles/ov2slam_node.dir/src/ov2slam_node.cpp.o: In function `main':
ov2slam_node.cpp:(.text.startup+0xd78): undefined reference to `cv::FileStorage::FileStorage(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
ov2slam_node.cpp:(.text.startup+0xf64): undefined reference to `cv::FileNode::string[abi:cxx11]() const'
ov2slam_node.cpp:(.text.startup+0xff8): undefined reference to `cv::FileNode::string[abi:cxx11]() const'
/home/thomas/catkin_ws/devel/lib/libov2slam.so: undefined reference to `cv::FileNode::operator float() const'
/home/thomas/catkin_ws/devel/lib/libov2slam.so: undefined reference to `cv::ORB::create(int, float, int, int, int, int, cv::ORB::ScoreType, int, int)'
/home/thomas/catkin_ws/devel/lib/libov2slam.so: undefined reference to `cv::FileNode::operator int() const'
/home/thomas/catkin_ws/devel/lib/libov2slam.so: undefined reference to `cv::FastFeatureDetector::create(int, bool, cv::FastFeatureDetector::DetectorType)'
/home/thomas/catkin_ws/devel/lib/libov2slam.so: undefined reference to `cv::FileNode::operator double() const'
/home/thomas/catkin_ws/devel/lib/libov2slam.so: undefined reference to `cv::Formatter::get(cv::Formatter::FormatType)'
collect2: error: ld returned 1 exit status
ov2slam/CMakeFiles/ov2slam_node.dir/build.make:316: recipe for target '/home/thomas/catkin_ws/devel/lib/ov2slam/ov2slam_node' failed
make[2]: *** [/home/thomas/catkin_ws/devel/lib/ov2slam/ov2slam_node] Error 1
CMakeFiles/Makefile2:510: recipe for target 'ov2slam/CMakeFiles/ov2slam_node.dir/all' failed
make[1]: *** [ov2slam/CMakeFiles/ov2slam_node.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed`

An idea where it could come from ?

ferreram, 2021-03-09:
Sounds like either you have two versions of OpenCV installed on your TX2 and their getting mixed up at compilation or something got wrong when OpenCV was installed.  You can try to force the use of OpenCV 3.2 (the version officially supported by ROS kinetic) in the CMakelists.txt : find_package(OpenCV 3.2).
Let me know if this resolves your issue.  
Also, if you want to use another version of OpenCV with ROS, you need to re-build the from source all the ROS packages that use OpenCV.

ferreram, 2021-03-11:
Here is some info about the use of another version of OpenCV than the native one coming with your ROS install : https://answers.ros.org/question/289090/how-to-specify-my-own-opencv3-for-cv_bridge/.

malicd, 2021-03-11:
Hi, just my two cents: the issue here is that you have most probably installed `ros-melodic-cv-bridge` via your package manager. By default it uses Opencv 3.2 and you have different OpenCV version on your system.
If you remove ros-melodic-cv-bridge` it and add [vision_opencv](https://github.com/ros-perception/vision_opencv) package to your project, you will compile it with your OpenCV version.
At least this is what solved the problem for me.

closed 2021-03-11 by neo-knight-td