ov2slam/ov2slam #10: Question about the Kitti Translation

Gatsby23, 2021-03-23:
Hi, Thanks for your excellent work. But there's some question about the test on kitti, the parameter file I used is the accurate parameter, but the result is too poor for this excellent work:
[image]
Here is my result, use the translation error tools from  https://github.com/LeoQLi/KITTI_odometry_evaluation_tool.
I think this is too low for this excellent work. Could you please help me to figure out ?

ferreram, 2021-03-23:
Which output file did you use to compute the result?  If you have used the "ov2slam_traj_kitti.txt" file, be aware that it stores the visual odometry poses of each image (i.e. it is the pose that is computed in the front-end of ov2slam for each image).
If you want to evaluate the obtained trajectory which takes into account the loop closures, you should either use "ov2slam_full_traj_wlc.txt" or "ov2slam_full_traj_wlc_opt.txt". 
You can convert these files to KITTI format using evo : https://github.com/MichaelGrupp/evo/wiki/Formats.

Also, are you using ov2slam with OpenGV?  If not, please install opengv as the results with OpenCV MVG functions might not be as accurate.

Gatsby23, 2021-03-23:
> Which output file did you use to compute the result? If you have used the "ov2slam_traj_kitti.txt" file, be aware that it stores the visual odometry poses of each image (i.e. it is the pose that is [...]

The result I used is ov2slam_traj_kitti.txt. I also install the OpenGV in my desktop. However, could you please test it ? Maybe there are some parameter I misunderstood and set the unreasonable value？

ferreram, 2021-03-23:
Running ov2slam on sequence 00 from KITTI I get the following result with https://github.com/LeoQLi/KITTI_odometry_evaluation_tool.  :
[image]

Gatsby23, 2021-03-23:
Thank you very much！Excellent Result！ Maybe you could put it on the readme to recommend others use this SLAM

closed 2021-03-23 by Gatsby23