ov2slam/ov2slam #9: Problem with the code

Gatsby23, 2021-03-23:
Hi, there are some problem with the launch, could you please help me to run the code.
The command when I run the code is:
```
 rosrun ov2slam ov2slam_node src/ov2slam/parameters_files/accurate/kitti/kitti_00-02.yaml 
```
and the kitti rosbag topic is:
```
/clock
/kitti/camera_color_left/camera_info
/kitti/camera_color_left/image_rect
/kitti/camera_color_right/camera_info
/kitti/camera_color_right/image_rect
/ov2slam_node/cam_pose_visual
/ov2slam_node/final_kfs_traj
/ov2slam_node/image_track
/ov2slam_node/kfs_traj
/ov2slam_node/local_kfs_window
/ov2slam_node/point_cloud
/ov2slam_node/vo_pose
/ov2slam_node/vo_traj
/rosout
/rosout_agg
/tf
```
But I met the error: 
```
OpenCV Error: Assertion failed (!fixedType() || ((Mat*)obj)->type() == mtype) in create, file /build/opencv-L2vuMj/opencv-3.2.0+dfsg/modules/core/src/matrix.cpp, line 2246
terminate called after throwing an instance of 'cv::Exception'
  what():  /build/opencv-L2vuMj/opencv-3.2.0+dfsg/modules/core/src/matrix.cpp:2246: error: (-215) !fixedType() || ((Mat*)obj)->type() == mtype in function create
```
Could you please give me some advice about that？

ferreram, 2021-03-23:
Hi,
Yes I know what's going on, I forgot to push the fix to handle RGB images.  I should be able to do this later today.
In the meantime, if you are in a hurry, you can change the following code in src/ov2slam_node.cpp:

```cpp

    cv::Mat getGrayImageFromMsg(const sensor_msgs::ImageConstPtr &img_msg)
    {
        // Get and prepare images
        cv_bridge::CvImageConstPtr ptr;
        try {    
            ptr = cv_bridge::toCvCopy(img_msg);
        } 
        catch(cv_bridge::Exception &e)
        {
            ROS_ERROR("\n\n\ncv_bridge exeception: %s\n\n\n", e.what());
        }

        if( ptr->image.channels() == 3 ) {
            cv::cvtColor(ptr->image, ptr->image, cv::COLOR_RGB2GRAY);
        }

        return ptr->image;
    }
```

into:

```cpp
    cv::Mat getGrayImageFromMsg(const sensor_msgs::ImageConstPtr &img_msg)
    {
        // Get and prepare images
        cv_bridge::CvImageConstPtr ptr;
        try {    
            ptr = cv_bridge::toCvCopy(img_msg);
        } 
        catch(cv_bridge::Exception &e)
        {
            ROS_ERROR("\n\n\ncv_bridge exeception: %s\n\n\n", e.what());
        }

        if( ptr->image.channels() == 3 ) {
            cv::Mat gray_img;
            cv::cvtColor(ptr->image, gray_img, cv::COLOR_RGB2GRAY);
            return gray_img;
        }

        return ptr->image;
    }
```

Gatsby23, 2021-03-23:
> Hi,
> Yes I know what's going on, I forgot to push the fix to handle RGB images. I should be able to do this later today.
> In the meantime, if you are in a hurry, you can change the following code in src/ov2slam_node.cpp:
> 
> ```c++
>     cv::Mat getGrayImageFromMsg(const sensor_msgs::ImageConstPtr &img_msg)
>     {
>         // Get and prepare images
>         cv_bridge::CvImageConstPtr ptr;
>         try {    
>             ptr = cv_bridge::toCvCopy(img_msg);
>         } 
>         catch(cv_bridge::Exception &e)
>         {
>             ROS_ERROR("\n\n\ncv_bridge exeception: %s\n\n\n", e.what());
>         }
> 
>         if( ptr->image.channels() == 3 ) {
>             cv::cvtColor(ptr->image, ptr->image, cv::COLOR_RGB2GRAY);
>         }
> 
>         return ptr->image;
>     }
> ```
> 
> into:
> 
> ```c++
>     cv::Mat getGrayImageFromMsg(const sensor_msgs::ImageConstPtr &img_msg)
>     {
>         // Get and prepare images
>         cv_bridge::CvImageConstPtr ptr;
>         try {    
>             ptr = cv_bridge::toCvCopy(img_msg);
>         } 
>         catch(cv_bridge::Exception &e)
>         {
>             ROS_ERROR("\n\n\ncv_bridge exeception: %s\n\n\n", e.what());
>         }
> 
>         if( ptr->image.channels() == 3 ) {
>             cv::Mat gray_img;
>             cv::cvtColor(ptr->image, gray_img, cv::COLOR_RGB2GRAY);
>             return gray_img;
>         }
> 
>         return ptr->image;
>     }
> ```
Thank you very much! It can run now, excellent work!

closed 2021-03-23 by ferreram