ov2slam/ov2slam #11: some detail question

HeadReaper-hc, 2021-03-28:
Hi,
     nice work!
     I have two question when I read the code as follows.

     [1]. visual_front_end.cpp line521,  **" if( nbkps < 8 )"**  why doesn't  "if(nbparallax < 8)"

     [2].ceres_parametrization.cpp line180  **"J_se3pose.block<2,3>(0,3).noalias() = J_lRcw * Sophus::SO3d::hat(wpt);"**
          I think **"J_se3pose.block<2,3>(0,3).noalias() = J_lRcw * Sophus::SO3d::hat(wpt-twc);"** is more accurate

ferreram, 2021-03-28:
Hi,
[1] OpenGV is expecting at least 8 samples to compute an Essential Matrix with RANSAC and Nister method.
[2] I think the jacobian is correct here.  I don't know how you derived your solution but be careful here as what we optimize is Twc (the pose that transforms a 3D vec from the camera frame to the world frame) but what we use in the reprojection error is Tcw, the inverse pose.

HeadReaper-hc, 2021-03-28:
[1]. visual_front_end.cpp line462 aready use **" if( nbkps < 8 )"**, so I think in line521 you want to use the number of 3d point to compute Essential Matrix, that' why I think here should use **"nbparallax"**
[2]. "Xc = Rwc.transpose() * Xw - Rwc.transpose() * twc", do you forget  the "Rwc.transpose()" before "twc" or just I made a mistake?

ferreram, 2021-03-28:
[1] Ah yes indeed, you're right.

[2] With the left SE(3) parametrization we have : (opt_Rwc, opt_twc) = (Exp(w) * Rwc, Exp(w) * twc + V(w) * p) and we optimize over w and p.  The Exp(w) will cancel out on twc because of the inverse (tcw = - Rcw * Exp(w)^-1 * (Exp(w) * twc + V(w) *p).  Hence, only Xw depends on w.

HeadReaper-hc, 2021-03-28:
Get it. Thanks!

closed 2021-03-28 by HeadReaper-hc

Alex-Beh, 2024-07-23:
@ferreram @HeadReaper-hc I am trying to understand the derivative of the parametrization. May I know what is the `V(w)` here?
> 
> [2] With the left SE(3) parametrization we have : (opt_Rwc, opt_twc) = (Exp(w) * Rwc, Exp(w) * twc + V(w) * p) and we optimize over w and p. The Exp(w) will cancel out on twc because of the inverse (tcw = - Rcw * Exp(w)^-1 * (Exp(w) * twc + V(w) *p). Hence, only Xw depends on w.

### left SE(3) parametrization
[image]

HeadReaper-hc, 2024-07-23:
@Alex-Beh 
 /// Group exponential
  ///
  /// This functions takes in an element of tangent space (= twist ``a``) and
  /// returns the corresponding element of the group SE(3).
  ///
  /// The first three components of ``a`` represent the translational part
  /// ``upsilon`` in the tangent space of SE(3), while the last three components
  /// of ``a`` represents the rotation vector ``omega``.
  /// To be more specific, this function computes ``expmat(hat(a))`` with
  /// ``expmat(.)`` being the matrix exponential and ``hat(.)`` the hat-operator
  /// of SE(3), see below.
  ///
  SOPHUS_FUNC static SE3<Scalar> exp(Tangent const& a) {
    using std::cos;
    using std::sin;
    Vector3<Scalar> const omega = a.template tail<3>();

    Scalar theta;
    SO3<Scalar> const so3 = SO3<Scalar>::expAndTheta(omega, &theta);
    Matrix3<Scalar> const V = SO3<Scalar>::leftJacobian(omega, theta);
    return SE3<Scalar>(so3, V * a.template head<3>());
  }

This is the code from Sophus Library, I think the 'V' in the code is corresponding to the V(w) in your question.
Not very sure, since I haven't seen anything related to SLAM for a long time.