ov2slam/ov2slam #12: terminate called after throwing an instance of std::out_of_range

kalfiq, 2021-04-06:
Hi, I am trying to run ov2slam with a camera. Initially, ov2slam can start with no problems, but it crashes when I make even the smallest movements. I am using a stereo camera and the frames are in grayscale. I hope you can guide me in rectifying this issue.

I have tried using the accurate, average, and fast parameter files.

Launching OV²SLAM...

Loading parameters file : /home/leanne/catkin_ws/src/ov2slam/parameters_files/accurate/kitti/kitti_00-02.yaml...

Parameters file loaded...

SLAM Parameters are being setup...

ROS visualizer is being created...

 SLAM Manager is being created...

 OPENCV CONTRIB NOT FOUND!  ORB DESCRIPTOR WILL BE USED!

 OPENGV FOUND!  OPENGV MVG FUNCTIONS WILL BE USED!

 SetupCalibration()

 Setting up camera, model selected : pinhole

Pinhole Camera Model created

 Camera Calibration set as : 

 K = 
698.185       0 641.895
      0 698.185 376.483
      0       0       1

 D =    0.176145   0.0290398  0.00101602 0.000205553
 opt K = 
[698.1849999999999, 0, 641.895;
 0, 698.1849999999999, 376.4835;
 0, 0, 1]
*********************************

Feature Extractor is constructed!

>>> Maximum nb of kps : 396
>>> Maximum kps dist : 35
>>> Minimum kps dist : 17
>>> Maximum kps qual : 0.001
>>> Minimum kps qual : 0.0005
*********************************

 Optimizer Object is created!

 Estimator Object is created!

 Optimizer Object is created!

 LoopCloser Object is created!

Mapper Object is created!

Mapper is ready to process Keyframes!

Sensors Grabber is created...

OV²SLAM is ready to process incoming images!

 Use LoopCLoser : 1
 LoopCloser is ready to process Keyframes!

 Estimator is ready to process Keyframes!

Starting the measurements reader thread!

=======================================================================
 BRIEF CANNOT BE USED ACCORDING TO CMAKELISTS (Opencv Contrib not enabled) 
 ORB WILL BE USED BriefDescriptorExtractor!  (BUT NO ROTATION  OR SCALE INVARIANCE ENABLED) 

=======================================================================
.
.
.
.
.

 - [Visual-Front-End]: Not ready to init yet!
 	>>> Init current parallax (1.48992 px)

 - [Visual-Front-End]: Not ready to init yet!
 	>>> Init current parallax (1.45592 px)

 - [Visual-Front-End]: Not ready to init yet!
 	>>> Init current parallax (0.313045 px)

 - [Visual-Front-End]: Not ready to init yet!
 	>>> Init current parallax (5.15133 px)

terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 423) >= this->size() (which is 396)
Aborted (core dumped)

Thank you! Apologies if it is basic question.

ferreram, 2021-04-06:
Hi,
My first guess is that you have not provided the right image size in the parameters file.
On what data are using OV2SLAM ? If you are using your own cameras, make sure that the cameras' parameters are right in the parameter file. Also, you are saying that you use a stereo camera but, according to the algo output, you are using it in monocular mode.

kalfiq, 2021-04-09:
Hi thanks for the reply. It's working now! The reason was exactly because I was running in monocular mode instead of stereo mode. I overlooked that setting in the parameters file. Many apologies!

closed 2021-04-09 by kalfiq