ov2slam/ov2slam #13: Crash after initialization

melhashash, 2021-04-06:
Hi, thanks for the work.

The program crashes after initialization. All dependencies are built with the `build_thirdparty.sh` script.
Built using catkin_make

**Here is the output:**

rosrun ov2slam ov2slam_node src/ov2slam/parameters_files/accurate/kitti/kitti_00-02.yaml 

Launching OV²SLAM...

Loading parameters file : src/ov2slam/parameters_files/accurate/kitti/kitti_00-02.yaml...

Parameters file loaded...

SLAM Parameters are being setup...

ROS visualizer is being created...

 SLAM Manager is being created...

 OPENCV CONTRIB FOUND!  BRIEF DESCRIPTOR WILL BE USED!

 OPENGV FOUND!  OPENGV MVG FUNCTIONS WILL BE USED!

 SetupCalibration()

 Setting up camera, model selected : pinhole

Pinhole Camera Model created

 Camera Calibration set as : 

 K = 
718.856       0 607.193
      0 718.856 185.216
      0       0       1

 D = 0 0 0 0
 opt K = 
[718.856, 0, 607.1928;
 0, 718.856, 185.2157;
 0, 0, 1]
*********************************

Feature Extractor is constructed!

>>> Maximum nb of kps : 396
>>> Maximum kps dist : 35
>>> Minimum kps dist : 17
>>> Maximum kps qual : 0.001
>>> Minimum kps qual : 0.0005
*********************************

 Optimizer Object is created!

 Estimator Object is created!

 Optimizer Object is created!

 LoopCloser Object is created!

Mapper Object is created!

Mapper is ready to process Keyframes!

Sensors Grabber is created...

OV²SLAM is ready to process incoming images!

 Estimator is ready to process Keyframes!

 Use LoopCLoser : 1
 LoopCloser is ready to process Keyframes!

Starting the measurements reader thread!

 	>>> Init current parallax (11.2527 px)

 - [Visual-Front-End]: Not ready to init yet!
 	>>> Init current parallax (21.1859 px)

 	>>> 5-pt EssentialMatrix Ransac :
 	>>> nb pts : 240 / avg. parallax : 24.3881 / nransac_iter_ : 100 / fransac_err_ : 3 / bdo_random : 1

 	>>> Epipolar nb outliers : 1
 	>>> Essential Mat init : -0.00208749 -0.00138448    0.249987
 	>>> Essential Mat Intialization run time : 11[ms]

WARNING: Logging before InitGoogleLogging() is written to STDERR
E0405 23:14:52.941429 17590 solver.cc:531] Terminating: Invalid configuration. Solver::Options::eta = 0. Violated constraint: Solver::Options::eta > 0.0
E0405 23:14:52.941587 17590 solver.cc:531] Terminating: Invalid configuration. Solver::Options::eta = 0. Violated constraint: Solver::Options::eta > 0.0
free(): invalid size
Aborted (core dumped)

ferreram, 2021-04-06:
Hi,
Could you provide more information about your setup?  What dataset are you using? What parameters are you using?

Shane922, 2021-04-06:
Hi, I have a similar problem:

Parameters file loaded...
SLAM Parameters are being setup...
......
......
 - [Visual-Front-End]: Not ready to init yet!
 	>>> Init current parallax (5.12937 px)

Segmentation fault (core dumped)

the command are as  followed:
roscore
rosrun ov2slam ov2slam_node my_path/catkin_ws/src/ov2slam/parameters_files/accurate/euroc/euroc_mono.yaml
rosbag play my_path/V1_01_easy.bag

note: I solved a problem like this as your suggestion [https://github.com/ov2slam/ov2slam/issues/4#issuecomment-790462813](url) and I disabled the opencv_contrib flag in CMakeLists by setting OFF.

ferreram, 2021-04-06:
Hi @Shane922 , 
Could you set the debug flag to "1" to see more specifically where the segfault is happening?  
My first guess is that the Ceres library that you are using was not compiled with "-march=native".  If you don't use OpenGV, you can remove the "-march=native" instruction from the CMakeLists.txt.  Otherwise, try to use the provided Ceres lib or re-compile your own Ceres lib with "-march=native".

melhashash, 2021-04-06:
> Hi,
> 
> Could you provide more information about your setup?  What dataset are you using? What parameters are you using?

I am using kitti sequence 00, monocular setup. I am using the accurate parameters with setting the flag for mono.
I have also tried with another dataset and faced the same problem.

melhashash, 2021-04-06:
I have uninstalled my previous version of ceres and only use the one comes with ov2slam.

ferreram, 2021-04-06:
Hi @melhashash ,
Could you set the debug flag to "1" in the parameters file and copy / paste here the output you get on the last 3-4 images before the crash?

melhashash, 2021-04-06:
Her

[debug_output.log](https://github.com/ov2slam/ov2slam/files/6265072/debug_output.log)

ferreram, 2021-04-06:
Thanks for sharing the logs.  I can see that something goes wrong when calling CeresPnP function but I don't know what.  Have you possibly changed anything else than the monocular mode in the parameters file?  If so, could you share your parameters file?

melhashash, 2021-04-06:
No, I am using the default parameters.

ferreram, 2021-04-06:
Ok that's weird.  What exactly is your setup (Ubuntu / ROS  version)?
Also, could you try removing the ov2slam folder in your catkin workspace build directory and run again the "build_thirdparty.sh" followed by catkin build?

melhashash, 2021-04-06:
Ubuntu 20.04
ROS noetic

melhashash, 2021-04-06:
BTW, I haven't dived into the code, would the CeresPnP be the first time to use Ceres in the pipeline or it was used after the initialization?

ferreram, 2021-04-06:
It is indeed.  That's why I have suggested to re-compile everything from scratch as something seems to be wrong with the version of Ceres that is linked to your ov2slam.

melhashash, 2021-04-06:
Just in case someone faces a similar issue. The problem was caused by conflicts in OpenCV versions, the one I manually built and the other shipped with ROS.

closed 2021-04-06 by melhashash

Shane922, 2021-04-07:
> Just in case someone faces a similar issue. The problem was caused by conflicts in OpenCV versions, the one I manually built and the other shipped with ROS.
Hi, @melhashash ,have you solved this problem? I am using Debian and ros-melodic with opencv 3.2.0(comes with ros). I also specified the version of opencv by "find_package(OpenCV 3.2.0 REQUIRED)" but it doesn't work. Could you tell me how you solved the problem, I'd appreciate it if you could help me.

melhashash, 2021-04-07:
> Hi, @melhashash ,have you solved this problem? I am using Debian and ros-melodic with opencv 3.2.0(comes with ros). I also specified the version of opencv by "find_package(OpenCV 3.2.0 REQUIRED)" [...]

I have ROS noetic that comes with OpenCV 4.2. Thus I got the same OpenCV version and built it from source to match ROS's version. So eventually, I didn't modify any cmake files.

I am sure there is a better way to handle this. Maybe the authors can tell us about the versions they used.

Shane922, 2021-04-07:
> Hi @Shane922 ,
> Could you set the debug flag to "1" to see more specifically where the segfault is happening?
> My first guess is that the Ceres library that you are using was not compiled with [...]

Thanks for your kind reply, I comment the "# set(CMAKE_CXX_FLAGS_RELEASE "-O3 -Wall -march=native")" in CMakeLists and re-catkin_make --pkg ov2slam,  but it still doesn't work. Here is the output_log:
[output.log](https://github.com/ov2slam/ov2slam/files/6268742/output.log)

Shane922, 2021-04-07:
Thanks very much for your help! @melhashash @ferreram 
I finally solved this problem just by specified the path of ceres_dir to the  Thirdparty in CMakelitsts.txt:
adding "set(Ceres_DIR ${PROJECT_SOURCE_DIR}/Thirdparty/ceres-solver/install/lib/cmake/Ceres)" before "find_package(Ceres REQUIRED)"