ov2slam/ov2slam #15: redefinition error

Vivek12199, 2021-04-12:
home/slam/catkin_ws/src/ov2slam/src/feature_extractor.cpp:44:9: error: redefinition of ‘class cv::ParallelLoopBodyLambdaWrapper’
   44 |   class ParallelLoopBodyLambdaWrapper : public ParallelLoopBody
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/opencv4/opencv2/core.hpp:3293,
                 from /home/slam/catkin_ws/src/ov2slam/include/frame.hpp:40,
                 from /home/slam/catkin_ws/src/ov2slam/include/feature_extractor.hpp:29,
                 from /home/slam/catkin_ws/src/ov2slam/src/feature_extractor.cpp:27:
/usr/include/opencv4/opencv2/core/utility.hpp:594:7: note: previous definition of ‘class cv::ParallelLoopBodyLambdaWrapper’
  594 | class ParallelLoopBodyLambdaWrapper : public ParallelLoopBody
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/slam/catkin_ws/src/ov2slam/src/feature_extractor.cpp:58:15: error: redefinition of ‘void cv::parallel_for_(const cv::Range&, std::function<void(const cv::Range&)>, double)’
   58 |   inline void parallel_for_(const cv::Range& range, std::function<void(const cv::Range&)> functor, double nstripes=-1.)
      |               ^~~~~~~~~~~~~
In file included from /usr/include/opencv4/opencv2/core.hpp:3293,
                 from /home/slam/catkin_ws/src/ov2slam/include/frame.hpp:40,
                 from /home/slam/catkin_ws/src/ov2slam/include/feature_extractor.hpp:29,
                 from /home/slam/catkin_ws/src/ov2slam/src/feature_extractor.cpp:27:
/usr/include/opencv4/opencv2/core/utility.hpp:609:13: note: ‘void cv::parallel_for_(const cv::Range&, std::function<void(const cv::Range&)>, double)’ previously defined here
  609 | inline void parallel_for_(const Range& range, std::function<void(const Range&)> functor, double nstripes=-1.)
      |             ^~~~~~~~~~~~~
make[2]: *** [ov2slam/CMakeFiles/ov2slam.dir/build.make:141: ov2slam/CMakeFiles/ov2slam.dir/src/feature_extractor.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1239: ov2slam/CMakeFiles/ov2slam.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j12 -l12" failed

ferreram, 2021-04-13:
You have to change "parameter_file.yaml" for one of the parameters file provided in the parameters_file folder.

closed 2021-04-22 by ferreram

FPSychotic, 2021-05-16:
hi, I change the name in the command by the one of a file, by my own fle name, i tried in different folders, but I get this error, m missing something very easy but no idea. thanks

iwire@iwire-UP-CHT01:~$ rosrun ov2slam ov2slam_node kitti_00-02.yaml

Launching OV²SLAM...

Loading parameters file : kitti_00-02.yaml...
Failed to open settings file...

ferreram, 2021-05-17:
@FPSychotic you need to specify the path to the parameter file as well.  For instance, if you open your terminal (or cd to) ov2slam directory, you can do: rosrun ov2slam ov2slam_node parameters_files/accurate/kitti/kitti_00-02.yaml.