ov2slam/ov2slam #17: 'catkin build' fails

MortenBeier, 2021-04-15:
Dear maintainers :)

I'm facing errors, when I try to 'catkin build':
1. I can't build as stated in the instructions. I have to go to '~/catkin_ws' and do a `catkin build ov2slam` from here (no big issue :))
2. The build process gives the error: "//usr/lib/aarch64-linux-gnu/libopencv_core.so.4.1: error adding symbols: DSO missing from command line" and I'm a little curious about the double slash in `//usr/lib/aarch64-linux-gnu.....` - is that OK?
[image]

I have tried to do a `catkin clean` and repeat the process with `catkin_make`, but I get a similar result.

I'm running on a Jetson Xavier NX and OpenCV comes as a part of Jetpack and when I look in '/usr/lib/aarch64-linux-gnu' I can see that most libs are in two versions (3 and 4): 
[image]

I have tried to link `libopencv_core.so` to `libopencv_core.so.3.2` instead of `libopencv_core.so.4.1`, as suggested here: [https://answers.opencv.org/question/217959/libopencv_coreso31-cannot-open-shared-object-file-no-such-file-or-directory/](url), but that doesn't change anything.

Do you have a tip or solution, which can bring me forward with this?

Thanks a lot in advance.

aeudes, 2021-04-15:
Ov²slam should be compiled with the same opencv lib ros is linked to.
You have two options :
- try to force ov²slam to opencv 3.2.0 in the cmake
with : 
```find_package(OpenCV 3.2.0 REQUIRED)```
But to be able to build, you will need to have the opencv3.2core-dev installed (and this package may conflict with the opencv4.0 version...).
A posible hack to solve that, is to temporarily disable nvidia repository in apt. make an update. Remove the opencv4.0-core-dev (it will force to uninstall ros package).
reinstall the opencv3.2-core-dev and the ros package. Reactivate the nvidia repository (be careful on future update then...).
An alternative here is to build opencv3.2 from source and force ov2slam to link with it.

- build ros from source with opencv4.0:
if you really need opencv4.0, the only way is to rebuild every ros package that depends on opencv from source (cvbridge, image_transport, vision_opencv, image_geometry,...). 
Then you could build ov²slam with the opencv4.0 version and you will not have conflicting version of the lib.

MortenBeier, 2021-04-25:
Thank you so much aeudes :)

I have now spend quite a lot of time to "build ros from source with opencv4.0" :-), ince I would prefer to keep OpenCV 4 / Melodic / Ubuntu 18.04, so I can also play around with the Nvidia AI demos.

I have followed [these](https://github.com/ros-perception/vision_opencv/issues/272)  - comment March 31,2020 by 'tkircher' instructions, and installed a number of additional libraries along the way and finally ROS and all 264 packages was built with OpenCV4 `./src/catkin/bin/catkin_make_isolated --install -DCMAKE_BUILD_TYPE=Release`

Unfortunately, I still get the same error:
[image]

Do you have an idea, what to do?
Should I somewhere specify the location of  OpenCV 4? (Other than what I have already described in the first post)? 
Should I disable or uninstall OpenCV 3 - and how?
I still have OpenCV 3 and 4 under `/usr/lib/aarch64-linux-gnu`.

ferreram, 2021-04-26:
Hi @MortenBeier,
You can try modifying in the CMakeLists.txt "find_package(OpenCV REQUIRED)" to "find_package(OpenCV 4 REQUIRED)" in order to force the use of OpenCV 4.
You can also add this line right after the find_package: "message( "OpenCV targeted version : ${OpenCV_VERSION}" )".  This will let you see what version of OpenCV is targeted when compiling OV2SLAM.

MortenBeier, 2021-04-26:
Hi @ferreram ,
The output looks valid:
```
-- Found OpenCV: /usr (found suitable version "4.1.1", minimum required is "4") 
OpenCV targeted version : 4.1.1

```
, but I get the same error when I `catkin_make --pkg ov2slam`

The full output looks like this:
[image]

aeudes, 2021-04-26:
You still have one dependancy of ov2slam build with opencv3:
the workspace still overlay /opt/ros/melodic, is it the correct path to your new install ?
did you force the opencv version on the thirdpary library obindex2 and ibowlcd ?

If it is not the thirdparty lib, you could found the faulty lib by using ldd. The list of lib linked by ov2slam should be in /home/morten/catkin_ws/build/ov2slam/CMakeFiles/ov2slam_node.dir/src/link.txt

MortenBeier, 2021-04-27:
Install path for ROS looks all right:
[image]

I have now changed the CMakeLists.txt for obindex2 and ibowlcd so they both contain `find_package(OpenCV 4 REQUIRED)` after that I have run `./build_thirdparty.sh` - but I still receive the error described previously.

I have attached /home/morten/catkin_ws/build/ov2slam/CMakeFiles/ov2slam_node.dir/src/link.txt to this message
[link.txt](https://github.com/ov2slam/ov2slam/files/6383238/link.txt)

As you might have detected this is somewhat new land to me, but I have tried to `ldd` a couple of the links in the links.txt file. I have attached three examples:
[image]
[image]
[image]

So it looks like libov2slam, libcv_bridge and probably more are linked to version OpenCV 3 AND 4. Is that the issue? And what can I do to solve that? would it be a solution to install conda and make an environment with ROS melodic, OpenCV3 and ov2slam only?

I have tried the following, but I get the same error
cd ~/catkin_ws/
catkin_make clean
catkin_make --pkg ov2slam

aeudes, 2021-04-27:
Are you sure the package in /opt/ros/melodic are the new one you have built ?
The libcv_bridge still depend on opencv3 that should not be if you built it from stratch with opencv4...

I think the default install path when you built with catkin_make_isolate is in install directory of the catkin workspace and not in /opt/ros/melodic (if you don't specify it with --install-space) : http://wiki.ros.org/Installation/Source
If so you have to change the ros workspace you extend with:
catkin config --extend thepathtotheinstalldirectory

bperiard, 2021-05-10:
I had a very similar issue on my Jetson Nano. It seemed like OpenCV was not linked by CMake. What made compile for me was to change ${OpenCV_LIBS} from private to public in target_link_libraries.
```
target_link_libraries(${PROJECT_NAME}
    PUBLIC
      ${catkin_LIBRARIES}
      Eigen3::Eigen
      Sophus::Sophus
      ${OpenCV_LIBS}
    PRIVATE
      Ceres::ceres
)
```

MortenBeier, 2021-05-12:
BAM! That made it for me. Thanks a lot @bperiard!!
I have just read about the use of  PUBLIC, PRIVATE and INTERFACE: https://cmake.org/pipermail/cmake/2016-May/063400.html - but how did you figure out the solution?

And thanks a lot also @aeudes  and @ferreram

bperiard, 2021-05-12:
@MortenBeier refer to the image of on your second post, I researched that part: 

libopencv_core.so.4.1: error adding symbols: DSO missing from command line

The following [post](https://github.com/kalov/ShapePFCN/issues/6) made me think that the problem was OpenCV was not linked by CMake. Then I tried what I previously explained.

But I am still not able to execute the code properly. I try to run the [benchmark script](https://github.com/ov2slam/ov2slam/blob/main/benchmark_scripts/euroc_bench.sh) that I adapted to my own folders, the euroc rosbag MH_01_easy.bag that I downloaded [here](https://projects.asl.ethz.ch/datasets/doku.php?id=kmavvisualinertialdatasets) seem to be running, but i can't visualise the outputs of the program.

@MortenBeier keep me updated if you are able to execute and visualise the output of the program.

ferreram, 2021-05-12:
Thanks @bperiard for providing your solution!  I will have to dig a bit this problem, we may eventually set OpenCV as PUBLIC. @MortenBeier glad to hear this solution worked for you!

@bperiard about your visualization problem, there is an rviz configuration file provided in the repository that you can directly load in rviz to watch the outputs.  Have you tried to use it?

bperiard, 2021-05-12:
@ferreram I finally got it to run thanks. I had to modify this line of the benchmark script :

`rosrun ov2slam ov2slam_node /home/maxime/catkin_ws/src/ov2slam/parameters_files/$param.yaml > log.txt &`

to this:

`rosrun ov2slam ov2slam_node /home/maxime/catkin_ws/src/ov2slam/parameters_files/fast/euroc/$param.yaml > log.txt &`

Then I had to update my version of RViz to the version compatible with ros-melodic as described [here](http://wiki.ros.org/rviz/UserGuide)

MortenBeier, 2021-05-15:
I just had time to run a brief test and run ` rosrun ov2slam ov2slam_node parameter_file.yaml` - it running fine, but when I view `ov2slam_visualization.rviz` nothing happens - except the time is running.  I will looking into @bperiard's changes.

I'm not sure if ov2SLAM by default is looking at a sample image set, and how to make it use my own camera (Global shutter stereo), I guess I have to specify that in the .yaml file, but I will look into that, when I'm back from holidays May 25 :-)

MortenBeier, 2021-05-15:
The original issue has been solved. Thanks a lot @aeudes , @ferreram  and @bperiard ! I will close the issue

closed 2021-05-15 by MortenBeier