ov2slam/ov2slam #19: I can't let ovslam run for more than three seconds

rainleong, 2021-04-24:
The previous compilation runs smoothly, but when I run the euroc dataset, there are problems, whether it is a binocular or a monocular node. After using the "rosbag play MH_03_medium.bag" command, it only ran on rviz for less than three seconds. Zhong, then the ovslam node told me that it has given up, the core has been dumped, and the initial point cloud picture appeared on the rviz leaf, and then there is no follow-up because of the ovslam node's crash. What is the problem?
[image]

ferreram, 2021-04-24:
Hi,
From what I see in your dump logs, either not all libraries are compiled with the "-march=native" flag (OV2SLAM, Ceres and OpenGV must all be compiled with or without this flag) or there is a mismatch between the Eigen lib linked to OV2SLAM, Ceres and OpenGV.
Please, first make sure that compilation is performed with or without the "-march=native" flag for all the aforementioned libs.  If that does not solve the problem, make sure that you are not linking these libs to different version of Eigen.

rainleong, 2021-04-26:
Hi,
Thank you for your answer. I re-compiled the relevant library and solved this problem. It should be that the previous openGV failed to compile and install. Now I have no problem recompiling and running. Thank you for your help. Happy life!

closed 2021-04-26 by ferreram