ov2slam/ov2slam #22: cv::Exception Because:m.dims >= 2 in function 'Mat'. I don't know the reason.

zheng-xiao, 2021-04-29:
terminate called after throwing an instance of ' Use LoopCLoser : 1
 LoopCloser is ready to process Keyframes!
cv::Exception'
  what(): OpenCV(3.4.11) /home/zxn/opencv-3.4.11/modules/core/src/matrix.cpp:406: error: (-215:Assertion failed) m.dims >= 2 in function 'Mat'

ferreram, 2021-04-29:
Could you please set the debug flag to 1 in the parameter file and copy paste what you get ?

zheng-xiao, 2021-04-30:
Thank you so much ferreram :)
This is what I get:

Parameters file loaded...

SLAM Parameters are being setup...

ROS visualizer is being created...

 SLAM Manager is being created...

 OPENCV CONTRIB FOUND!  BRIEF DESCRIPTOR WILL BE USED!

 OPENGV NOT FOUND!  OPENCV MVG FUNCTIONS WILL BE USED!

 SetupCalibration()

 Setting up camera, model selected : pinhole

Pinhole Camera Model created

 Camera Calibration set as : 

 K = 
1961.31       0 613.414
      0  1962.2  290.51
      0       0       1

 D =   -0.556723    0.435529 -0.00566285  0.00018195
 opt K = 
[1961.314378211657, 0, 613.4136655058605;
 0, 1962.195285830189, 290.5095360265245;
 0, 0, 1]
 Setting up camera, model selected : pinhole

Pinhole Camera Model created

 Camera Calibration set as : 

 K = 
1969.68       0 610.353
      0 1970.34 255.043
      0       0       1

 D =  -0.546421   0.313459 -0.0050466 0.00158827
 opt K = 
[1969.676069548325, 0, 610.352836541342;
 0, 1970.337570057917, 255.0429704379117;
 0, 0, 1]
 Camera Extrinsic : 

 Rc0ci = 
1 0 0
0 1 0
0 0 1

 tc0ci = 0.195059        0        0
 SetupStereoCalibration()

 Alpha : 0
 Kl : 
[1961.314378211657, 0, 613.4136655058605;
 0, 1962.195285830189, 290.5095360265245;
 0, 0, 1]
 Kr : 
[1969.676069548325, 0, 610.352836541342;
 0, 1970.337570057917, 255.0429704379117;
 0, 0, 1]
 Dl : 
[-0.5567227539749868;
 0.4355288654041437;
 -0.005662847214116913;
 0.0001819504516358611]
 Dr : 
[-0.5464212457621284;
 0.3134593275348545;
 -0.00504660398179259;
 0.001588267986869001]
 Rl : 
[1, 0, 0;
 0, 1, 0;
 0, 0, 1]
 Rr : 
[1, 0, 0;
 0, 1, 0;
 0, 0, 1]
 Pl : 
[2046.35420981104, 0, 606.1313171386719, 0;
 0, 2046.35420981104, 257.6530113220215, 0;
 0, 0, 1, 0]
 Pr : 
[2046.35420981104, 0, 606.1313171386719, -399.1602355459158;
 0, 2046.35420981104, 257.6530113220215, 0;
 0, 0, 1, 0]

Computing the stereo rectification mapping!

 Undist+Rect Camera Calibration set as : 

 K = 
2046.35       0 606.131
      0 2046.35 257.653
      0       0       1

 D = 0 0 0 0

 ROI = [1275 x 715 from (5, 5)]

Computing the stereo rectification mapping!

 Undist+Rect Camera Calibration set as : 

 K = 
2046.35       0 606.131
      0 2046.35 257.653
      0       0       1

 D = 0 0 0 0

 ROI = [1275 x 715 from (5, 5)]
*********************************

Feature Extractor is constructed!

>>> Maximum nb of kps : 777
>>> Maximum kps dist : 35
>>> Minimum kps dist : 17
>>> Maximum kps qual : 0.001
>>> Minimum kps qual : 0.0005
*********************************

 Optimizer Object is created!

 Estimator Object is created!

 Optimizer Object is created!

 LoopCloser Object is created!

Mapper Object is created!

Mapper is ready to process Keyframes!

 Use LoopCLoser : 1
 LoopCloser is ready to process Keyframes!

 Estimator is ready to process Keyframes!
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(3.4.11) /home/zxn/opencv-3.4.11/modules/core/src/matrix.cpp:406: error: (-215:Assertion failed) m.dims >= 2 in function 'Mat'

ferreram, 2021-04-30:
Thanks for sharing what you get.  Can you copy paste what you get using gdb and backtracing?

zheng-xiao, 2021-04-30:
Hi, Ferreram, thanks for answer my question. But I don't know what is gdb and backtracing.

ferreram, 2021-04-30:
gdb is a debugging tool.  To install it: "sudo apt install gdb".
Then to use it: 
```
gdb ~/catkin_ws/devel/lib/ov2slam/ov2slam_node
r your_path/parameter_file.yaml
bt
```

zheng-xiao, 2021-05-01:
Hi, Ferreram, thanks for answer my question. I find my mistake.  I accidentally added a cv::Mat in the ov2slam.cpp.

closed 2021-05-03 by ferreram

Shawn-MY, 2024-04-11:
Hello, I have the same problem, how do you solve it?