ov2slam/ov2slam #27: Segmentation fault

dulex123, 2021-06-30:
When I try to launch ov2slam_node the following error appears:

```
Launching OV²SLAM...

Loading parameters file : src/ov2slam/parameters_files/wing/wing.yaml...
Stack trace (most recent call last):
#2    Object "/lib/aarch64-linux-gnu/libc.so.6", at 0x7f7f1c757b, in __vsnprintf_chk
#1    Object "/lib/aarch64-linux-gnu/libc.so.6", at 0x7f7f130f37, in vfprintf
#0    Object "/lib/aarch64-linux-gnu/libc.so.6", at 0x7f7f164450, in strlen
Segmentation fault (Address not mapped to object [0x65646f4e646461])
Segmentation fault (core dumped)
```

I have used -march=native flag for OpenGV and used the bundled ceres.

ferreram, 2021-06-30:
Hi,
Is the same happening if you run it with one of the example files (e.g. rosrun ov2slam ov2slam_node your_path/ov2slam/parameters_file/accurate/euroc/euroc_stereo.yaml) ?
Also, is this happening directly when you run the node or right after you start playing your bag file?

dulex123, 2021-07-01:
It happens immediately after running the program. I have already tried some of the default parameter files but it happens also. Is there any way to inspect where it happens?

ferreram, 2021-07-01:
Could you try running it with gdb and backtrace the segfault?

closed 2021-08-03 by ferreram