ov2slam/ov2slam #31: Stop running halfway when running on the TUM dataset

xtcl, 2021-11-02:
I have run ov2slam on KITTI, EUROC and my own camera, and they all work well. 
**But when I tried it on the TUM data set, it always terminates in the middle of the TUM bag playback(and sometimes the running time was longer and sometimes shorter), and then output Stack trace, The tail of the terminal output is as follows:**
```bash
 >>> 2.BA_Optimize : 5.31645 ms 

iter      cost      cost_change  |gradient|   |step|    tr_ratio  tr_radius  ls_iter  iter_time  total_time
   0  1.391902e+03    0.00e+00    8.82e+02   0.00e+00   0.00e+00  1.00e+04        0    9.23e-04    1.91e-03
   1  1.390386e+03    1.52e+00    7.34e+00   1.95e-02   1.02e+00  3.00e+04        1    2.05e-03    3.96e-03

Solver Summary (v 2.0.0-eigen-(3.3.4)-lapack-suitesparse-(5.1.2)-cxsparse-(3.1.9)-eigensparse-no_openmp)

                                     Original                  Reduced
Parameter blocks                          469                      399
Parameters                                904                      423
Effective parameters                      832                      419
Residual blocks                          2339                     2339
Residuals                                4678                     4678

Minimizer                        TRUST_REGION

Sparse linear algebra library    SUITE_SPARSE
Trust region strategy     LEVENBERG_MARQUARDT

                                        Given                     Used
Linear solver                    SPARSE_SCHUR             SPARSE_SCHUR
Threads                                     1                        1
Linear solver ordering                  396,4                    395,4
Schur structure                         2,1,6                    2,d,d

Cost:
Initial                          1.391902e+03
Final                            1.390386e+03
Change                           1.516119e+00

Minimizer iterations                        2
Successful steps                            2
Unsuccessful steps                          0

Time (in seconds):
Preprocessor                         0.000984

  Residual only evaluation           0.000901 (2)
  Jacobian & residual evaluation     0.001652 (2)
  Linear solver                      0.001165 (2)
Minimizer                            0.004041

Postprocessor                        0.000032
Total                                0.005057

Termination:                      CONVERGENCE (Function tolerance reached. |cost_change|/cost: 1.134214e-05 <= 1.000000e-03)

terminate called without an active exception
Stack trace (most recent call last):
#8    Object "", at 0xffffffffffffffff, in 
#7    Object "/home/xtcl/ros_ws/devel/lib/ov2slam/ov2slam_node", at 0x55c049cca6a9, in _start
#6    Source "../csu/libc-start.c", line 310, in __libc_start_main [0x7f2326852bf6]
#5  | Source "/home/xtcl/ros_ws/src/ov2slam/src/ov2slam_node.cpp", line 208, in ~thread
    |   207:     // Start a thread for providing new measurements to the SLAM
    | > 208:     std::thread sync_thread(&SensorsGrabber::sync_process, &sb);
    |   209: 
    |   210:     // ROS Spin
      Source "/usr/include/c++/7/thread", line 135, in main [0x55c049cca345]
        132:     ~thread()
        133:     {
        134:       if (joinable())
      > 135: 	std::terminate();
        136:     }
        137: 
        138:     thread& operator=(const thread&) = delete;
#4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.25", at 0x7f2326eccb20, in std::terminate()
#3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.25", at 0x7f2326eccae5, in 
#2    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.25", at 0x7f2326ec6956, in 
#1    Source "/build/glibc-S9d2JN/glibc-2.27/stdlib/abort.c", line 79, in abort [0x7f2326871920]
#0    Source "../sysdeps/unix/sysv/linux/raise.c", line 51, in raise [0x7f232686ffb7]
Aborted (Signal sent by tkill() 32518 1000)
```
**The parameter file of TUM is as follows:**
```yaml
%YAML 1.0
---
#--------------------------------------------------------------------------------------------
# Camera Parameters. Adjust them!
#--------------------------------------------------------------------------------------------

Camera.topic_left: /camera/rgb/image_color

Camera.model_left: pinhole

Camera.left_nwidth: 640
Camera.left_nheight: 480

# Camera calibration and distortion parameters (OpenCV) 

Camera.fxl: 517.3
Camera.fyl: 516.5
Camera.cxl: 318.6
Camera.cyl: 255.3

Camera.k1l: 0.2624
Camera.k2l: -0.9531
Camera.p1l: -0.0054
Camera.p2l: 0.0026

# Camera Extrinsic parameters T_b_ci ( v_b = T_b_ci * v_ci )
body_T_cam0: !!opencv-matrix
   rows: 4
   cols: 4
   dt: d
   data: [1., 0., 0., 0.,
          0., 1., 0., 0.,
          0., 0., 1., 0.,
           0., 0., 0., 1.]

#--------------------------------------------------------------------------------------------
# SLAM Parameters. Adjust them!
#--------------------------------------------------------------------------------------------
debug: 1
log_timings: 1

mono: 1
stereo: 0

force_realtime: 0

# Estimator Mode
slam_mode: 1

# Loop Closing mode
buse_loop_closer: 0

# Stereo rectify or not
bdo_stereo_rect: 0
alpha: 0.

# Undist Image or not
bdo_undist: 0

# Init / KF px th.
finit_parallax: 20.

# Features Extractor
use_shi_tomasi: 0
use_fast: 0
use_brief: 1
use_singlescale_detector: 1

# Min dist between kps (define the number of kps)
nmaxdist: 25 #30

# Features quality th.
nfast_th: 10
dmaxquality: 0.001 # used for gftt or singlescale

# Pre-processing
use_clahe: 1
fclahe_val: 3

# KLT related settings.
do_klt: 1
klt_use_prior: 1
btrack_keyframetoframe: 0
nklt_win_size: 9
nklt_pyr_lvl: 3

# Opencv Default: 30 iters. + 0.01 prec.
nmax_iter: 30
fmax_px_precision: 0.01

fmax_fbklt_dist: 0.5
nklt_err: 30.

# Matching th.
bdo_track_localmap: 1

fmax_desc_dist: 0.2 #0.35 # Ratio from desc size (for binary desc.) 
fmax_proj_pxdist: 2.

# Randomize RANSAC?
doepipolar: 1 #1
dop3p : 1
bdo_random: 1
nransac_iter: 100
fransac_err: 3.

# Reproj err th.
fmax_reproj_err: 3.
buse_inv_depth: 1

# Ceres related parameters
robust_mono_th: 5.9915 # (20% : 3.2189 / 10% :  4.6052 / 5% : 5.9915 / 2%: 7.8240 / 1%: 9.2103)  

use_sparse_schur: 1
use_dogleg: 0
use_subspace_dogleg: 0
use_nonmonotic_step: 0

# Estimator parameters
apply_l2_after_robust: 1

nmin_covscore: 25 # Min co-obs for optimizing in SLAM mode BA

# Map Filtering
fkf_filtering_ratio: 0.95

# Final Pass
do_full_ba: 0 #0

```

**When kitti and euroc are running, it can run until the end of bag playback, but the same Stack trace will be output at the end：**
```bash
 Estimator thread is exiting.

 Bag reader SyncProcess thread is terminating!

Mapper is stopping!

 Going to write the computed trajectory into : ov2slam_traj.txt

Trajectory file written!

 Going to write the computed trajectory into : ov2slam_traj_kitti.txt

KITTITrajectory file written!

 Going to write the computed KFs trajectory into : ov2slam_kfs_traj.txt

Kfs Trajectory file written!

OV²SLAM is stopping!
terminate called without an active exception
Stack trace (most recent call last):
#8    Object "", at 0xffffffffffffffff, in 
#7    Object "/home/xtcl/ros_ws/devel/lib/ov2slam/ov2slam_node", at 0x559a113976a9, in _start
#6    Source "../csu/libc-start.c", line 310, in __libc_start_main [0x7f4f1a5d0bf6]
#5  | Source "/home/xtcl/ros_ws/src/ov2slam/src/ov2slam_node.cpp", line 208, in ~thread
    |   207:     // Start a thread for providing new measurements to the SLAM
    | > 208:     std::thread sync_thread(&SensorsGrabber::sync_process, &sb);
    |   209: 
    |   210:     // ROS Spin
      Source "/usr/include/c++/7/thread", line 135, in main [0x559a11397345]
        132:     ~thread()
        133:     {
        134:       if (joinable())
      > 135: 	std::terminate();
        136:     }
        137: 
        138:     thread& operator=(const thread&) = delete;
#4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.25", at 0x7f4f1ac4ab20, in std::terminate()
#3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.25", at 0x7f4f1ac4aae5, in 
#2    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.25", at 0x7f4f1ac44956, in 
#1    Source "/build/glibc-S9d2JN/glibc-2.27/stdlib/abort.c", line 79, in abort [0x7f4f1a5ef920]
#0    Source "../sysdeps/unix/sysv/linux/raise.c", line 51, in raise [0x7f4f1a5edfb7]
Aborted (Signal sent by tkill() 7465 1000)
```

aeudes, 2021-11-06:
Hello,
this is caused by the sync_thread not properly join before ov2slam exit. This should not cause any other problem than trigger the backtrace. Adding a sync_thread.join() before return 0; in ov2slam_node.cpp should probably fix it.
Thanks for reporting, I'll add a fix !

Did you manage to fix your problem with the TUM dataset as it seam to me that this bug is unrelated ?
If not, you could try to disable, the auto-detection of the end of sequence by changing :
l.213 in ov2slam.cpp from:
```
 if( c1 && c2 && c3 )
```
to:
```
 if( ros::isShuttingDown() )
```
To wait a user request ending than an end of images sequences.

We already have some trouble with this part earlier when image time stamp are messy but we think it have been fixed...

xtcl, 2021-11-11:
Hi @aeudes 
I have solved the problem of running interruption on the TUM dataset, by using my own generated rosbag from the image instead of the official bag file.
As you said, this Stack trace about threading issues does not cause any problems, it just doesn't look neat.
Following your suggestion, I tried adding "sync_thread.join();" before "return 0;", but it didn't solve the problem, and then I further added "slamthread.join();", this time It works, and no Stack trace is output, but this brings the problem of repeatedly releasing memory. The original problem has been solved, thank you for your help.

closed 2021-11-11 by xtcl

komalbagai-monarch, 2023-06-22:
Hello, I am getting the same error with KITTI sequence 07 bagfile. Can anyone help me out?

Thanks!

aeudes, 2023-06-22:
Hello, if you get this error exactly at the end of the bag, you could safely ignore it.

komalbagai-monarch, 2023-06-23:
Thank you for such a prompt response! Yes, I was ignoring it but I wanted to save the trajectories for offline use.
Do you recommend any easy way to do that? It exists before it can save it.