ov2slam/ov2slam #34: how install

DandyLorenz, 2021-11-30:
I flow this step run on ubuntu 16.04 ,i installed ros kinetic,
1. Prerequisites Make sure that the pcl_ros package is installed :

    sudo apt install ros-distro-pcl-ros
or even
    rosdep install ov2slam

How to install  ros-distro-pcl-ros  and   ov2slam，because I am not found ppa source

ferreram, 2021-11-30:
Hi,

You should first install ROS kinetic (following this tutorial : http://wiki.ros.org/kinetic/Installation/Ubuntu).  Then install PCL for ROS with the command: "sudo apt install ros-distro-pcl-ros".

Once this done, you can install OV2SLAM by following the step from section "2. Installation" on the main page of the repository.

Best,
Maxime

aeudes, 2021-11-30:
BTW: distro is a placeholder for your ros distribution for kinetic it is :
```
sudo apt install ros-kinetic-pcl-ros
```

poudyalbot, 2021-12-01:
Res sir,
Will the ov2slam will have tight  imu fusion in the future.
Thank you

DandyLorenz, 2021-12-01:
Thank you for your warm answer. How can I execute the following commands? 
```
cd /root/catkin_ws/
rosdep install ov2slam
```
My execution result is invalid.
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ov2slam: Cannot locate rosdep definition for [catkin]
```

ferreram, 2021-12-06:
@DandyLorenz  You can skip the rosdep install.  Just make sure to run "sudo apt install ros-kinetic-pcl-ros" as pointed out by Aeudes.  Then, build the thirdparty lib and you should finally be able to build ov2slam.

closed 2021-12-07 by DandyLorenz