ov2slam/ov2slam #35: catkin_make --pkg ov2slam error: recipe for target 'ov2slam/CMakeFiles/ov2slam.dir/src/visual_front_end.cpp.o' failed

DandyLorenz, 2021-12-01:
Environment description：ubuntu16.04 installed ROS kinetic,I successed run ORB-SLAM2，I am find great ov2slam。
I installed 
- opencv 3.4.3
- gcc (Ubuntu 5.4.0-6ubuntu1~16.04.12) 5.4.0 20160609
- eigen3 eigen-3.3.7
- Pongolin Pangolin-0.5 (Not the latest)
     
                 
### I follow the instructions to install
**1.init ROS and workspace**
    I installed ROS kinetic,and rviz also installed,I Make sure i installed  follow this conmand:  sudo apt install ros-kinetic-pcl-ros
```
mkdir -p /root/catkin_ov2slam/src && cd /root/catkin_ov2slam/src
    catkin_init_workspace
    cd ../ && catkin_make 
```
I run conmmand: **rosdep install ov2slam**, it warn me :
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ov2slam: Cannot locate rosdep definition for [catkin]
```
I do this conmmand ingonre step: 
`rosdep install --from-paths /root/catkin_ov2slam/src --ignore-src -y --rosdistro=kinetic`

**2. Installation**
2.0 Clone
Clone the git repository in your catkin workspace:
2.1 Build Thirdparty libs and also build opengv
2.2 Build OV²SLAM
Build OV²SLAM package with :
    cd ~/catkin_ov2slam/
    **catkin_make --pkg ov2slam**（report errors）
    _source ~/catkin_ov2slam/devel/setup.bash (Not yet implemented here)_

Problem description：After I have completed the necessary preparations,I run catkin_make --pkg ov2slam error，
some report  errors:

```
In file included from /root/catkin_ov2slam/src/ov2slam/include/frame.hpp:42:0,
                 from /root/catkin_ov2slam/src/ov2slam/src/frame.cpp:27:
/root/catkin_ov2slam/src/ov2slam/Thirdparty/Sophus/sophus/se3.hpp: In instantiation of ‘Sophus::SE3<typename Eigen::internal::traits<T>::Scalar> Sophus::SE3Base<Derived>::inverse() const [with Derived = Sophus::SE3<double>; typename Eigen::internal::traits<T>::Scalar = double]’:
/root/catkin_ov2slam/src/ov2slam/src/frame.cpp:748:24:   required from here
/root/catkin_ov2slam/src/ov2slam/Thirdparty/Sophus/sophus/se3.hpp:210:35: error: no match for ‘operator*’ (operand types are ‘Sophus::SO3<double>’ and ‘const ScalarMultipleReturnType {aka const Eigen::CwiseUnaryOp<Eigen::internal::scalar_multiple_op<double>, const Eigen::Matrix<double, 3, 1> >}’)
     return SE3<Scalar>(invR, invR * (translation() * Scalar(-1)));
ov2slam/CMakeFiles/ov2slam.dir/build.make:143: recipe for target 'ov2slam/CMakeFiles/ov2slam.dir/src/feature_extractor.cpp.o' failed
make[2]: *** [ov2slam/CMakeFiles/ov2slam.dir/src/feature_extractor.cpp.o] Error 1
ov2slam/CMakeFiles/ov2slam.dir/build.make:130: recipe for target 'ov2slam/CMakeFiles/ov2slam.dir/src/camera_calibration.cpp.o' failed
make[2]: *** [ov2slam/CMakeFiles/ov2slam.dir/src/camera_calibration.cpp.o] Error 1
ov2slam/CMakeFiles/ov2slam.dir/build.make:117: recipe for target 'ov2slam/CMakeFiles/ov2slam.dir/src/slam_params.cpp.o' failed
make[2]: *** [ov2slam/CMakeFiles/ov2slam.dir/src/slam_params.cpp.o] Error 1
In file included from /root/catkin_ov2slam/src/ov2slam/Thirdparty/Sophus/sophus/so3.hpp:8:0,
                 from /root/catkin_ov2slam/src/ov2slam/Thirdparty/Sophus/sophus/se3.hpp:7,
                 from /root/catkin_ov2slam/src/ov2slam/include/camera_visualizer.hpp:37,
                 from /root/catkin_ov2slam/src/ov2slam/src/camera_visualizer.cpp:31:
/root/catkin_ov2slam/src/ov2slam/Thirdparty/Sophus/sophus/so2.hpp:106:40: error: ‘ScalarBinaryOpTraits’ in namespace ‘Eigen’ does not name a template type
   using ReturnScalar = typename Eigen::ScalarBinaryOpTraits<
                                        ^
/root/catkin_ov2slam/src/ov2slam/Thirdparty/Sophus/sophus/so2.hpp:110:26: error: ‘ReturnScalar’ was not declared in this scope
   using SO2Product = SO2<ReturnScalar<OtherDerived>>;
                          ^
/root/catkin_ov2slam/src/ov2slam/Thirdparty/Sophus/sophus/so2.hpp:110:39: error: template argument 1 is invalid
   using SO2Product = SO2<ReturnScalar<OtherDerived>>;
                                       ^
/root/catkin_ov2slam/src/ov2slam/Thirdparty/Sophus/sophus/so2.hpp:113:32: error: ‘ReturnScalar’ was not declared in this scope
   using PointProduct = Vector2<ReturnScalar<PointDerived>>;
                                ^
/root/catkin_ov2slam/src/ov2slam/Thirdparty/Sophus/sophus/so2.hpp:113:45: error: template argument 1 is invalid
   using PointProduct = Vector2<ReturnScalar<PointDerived>>;
                                             ^
[... 391 lines cut ...]
                 std::is_same<Scalar, ReturnScalar<OtherDerived>>::value>::type>
                                                               ^
/root/catkin_ov2slam/src/ov2slam/Thirdparty/Sophus/sophus/se3.hpp:359:75: error: ‘type’ in namespace ‘::’ does not name a type
                 std::is_same<Scalar, ReturnScalar<OtherDerived>>::value>::type>
                                                                           ^
In file included from /usr/local/include/boost/detail/endian.hpp:9:0,
                 from /usr/include/pcl-1.7/pcl/PCLPointCloud2.h:11,
                 from /usr/include/pcl-1.7/pcl/conversions.h:48,
                 from /opt/ros/kinetic/include/pcl_ros/point_cloud.h:8,
                 from /root/catkin_ov2slam/src/ov2slam/include/map_manager.hpp:32,
                 from /root/catkin_ov2slam/src/ov2slam/include/visual_front_end.hpp:35,
                 from /root/catkin_ov2slam/src/ov2slam/src/visual_front_end.cpp:29:

/usr/include/eigen3/Eigen/src/Core/../plugins/CommonCwiseUnaryOps.h:83:1: note:   no known conversion for argument 1 from ‘Sophus::SO3<double>’ to ‘const Scalar& {aka const double&}’
ov2slam/CMakeFiles/ov2slam.dir/build.make:91: recipe for target 'ov2slam/CMakeFiles/ov2slam.dir/src/visual_front_end.cpp.o' failed
make[2]: *** [ov2slam/CMakeFiles/ov2slam.dir/src/visual_front_end.cpp.o] Error 1
ov2slam/CMakeFiles/ov2slam.dir/build.make:65: recipe for target 'ov2slam/CMakeFiles/ov2slam.dir/src/ov2slam.cpp.o' failed
make[2]: *** [ov2slam/CMakeFiles/ov2slam.dir/src/ov2slam.cpp.o] Error 1
CMakeFiles/Makefile2:488: recipe for target 'ov2slam/CMakeFiles/ov2slam.dir/all' failed
make[1]: *** [ov2slam/CMakeFiles/ov2slam.dir/all] Error 2
Makefile:143: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
```
I hope this problem is a typical error and can be effectively solved，thanks dear all~

DandyLorenz, 2021-12-01:
Supplementary note: the first error after operation is as follows：    **catkin_make --pkg ov2slam**
I guess it's about ‘Eigen3’ problem.
```

In file included from /root/catkin_ov2slam/src/ov2slam/Thirdparty/Sophus/sophus/se3.hpp:7:0,
                 from /root/catkin_ov2slam/src/ov2slam/include/multi_view_geometry.hpp:33,
                 from /root/catkin_ov2slam/src/ov2slam/src/feature_tracker.cpp:33:
/root/catkin_ov2slam/src/ov2slam/Thirdparty/Sophus/sophus/so3.hpp:110:40: error: ‘ScalarBinaryOpTraits’ in namespace ‘Eigen’ does not name a template type
   using ReturnScalar = typename Eigen::ScalarBinaryOpTraits<
                                        ^
/root/catkin_ov2slam/src/ov2slam/Thirdparty/Sophus/sophus/so3.hpp:114:26: error: ‘ReturnScalar’ was not declared in this scope
   using SO3Product = SO3<ReturnScalar<OtherDerived>>;
                          ^
/root/catkin_ov2slam/src/ov2slam/Thirdparty/Sophus/sophus/so3.hpp:114:39: error: template argument 1 is invalid
   using SO3Product = SO3<ReturnScalar<OtherDerived>>;
                                       ^
/root/catkin_ov2slam/src/ov2slam/Thirdparty/Sophus/sophus/so3.hpp:117:32: error: ‘ReturnScalar’ was not declared in this scope
   using PointProduct = Vector3<ReturnScalar<PointDerived>>;
                                ^
/root/catkin_ov2slam/src/ov2slam/Thirdparty/Sophus/sophus/so3.hpp:117:45: error: template argument 1 is invalid
   using PointProduct = Vector3<ReturnScalar<PointDerived>>;
                                             ^
/root/catkin_ov2slam/src/ov2slam/Thirdparty/Sophus/sophus/so3.hpp:120:43: error: ‘ReturnScalar’ was not declared in this scope
   using HomogeneousPointProduct = Vector4<ReturnScalar<HPointDerived>>;
                                           ^
/root/catkin_ov2slam/src/ov2slam/Thirdparty/Sophus/sophus/so3.hpp:120:56: error: template argument 1 is invalid
   using HomogeneousPointProduct = Vector4<ReturnScalar<HPointDerived>>;
                                                        ^
/root/catkin_ov2slam/src/ov2slam/Thirdparty/Sophus/sophus/so3.hpp:329:15: error: ‘SO3Product’ does not name a type
   SOPHUS_FUNC SO3Product<OtherDerived> operator*(
               ^
/root/catkin_ov2slam/src/ov2slam/Thirdparty/Sophus/sophus/so3.hpp:362:15: error: ‘PointProduct’ does not name a type
   SOPHUS_FUNC PointProduct<PointDerived> operator*(
               ^
/root/catkin_ov2slam/src/ov2slam/Thirdparty/Sophus/sophus/so3.hpp:377:15: error: ‘HomogeneousPointProduct’ does not name a type
   SOPHUS_FUNC HomogeneousPointProduct<HPointDerived> operator*(
               ^
/root/catkin_ov2slam/src/ov2slam/Thirdparty/Sophus/sophus/so3.hpp:399:38: error: ‘ReturnScalar’ was not declared in this scope
                 std::is_same<Scalar, ReturnScalar<OtherDerived>>::value>::type>
                                      ^
/root/catkin_ov2slam/src/ov2slam/Thirdparty/Sophus/sophus/so3.hpp:399:51: error: template argument 2 is invalid
                 std::is_same<Scalar, ReturnScalar<OtherDerived>>::value>::type>
                                                   ^
/root/catkin_ov2slam/src/ov2slam/Thirdparty/Sophus/sophus/so3.hpp:399:63: error: template argument 1 is invalid
                 std::is_same<Scalar, ReturnScalar<OtherDerived>>::value>::type>
                                                               ^
/root/catkin_ov2slam/src/ov2slam/Thirdparty/Sophus/sophus/so3.hpp:399:75: error: ‘type’ in namespace ‘::’ does not name a type
                 std::is_same<Scalar, ReturnScalar<OtherDerived>>::value>::type>
                                                                           ^
In file included from /root/catkin_ov2slam/src/ov2slam/include/multi_view_geometry.hpp:33:0,
                 from /root/catkin_ov2slam/src/ov2slam/src/feature_tracker.cpp:33:
/root/catkin_ov2slam/src/ov2slam/Thirdparty/Sophus/sophus/se3.hpp:85:40: error: ‘ScalarBinaryOpTraits’ in namespace ‘Eigen’ does not name a template type
   using ReturnScalar = typename Eigen::ScalarBinaryOpTraits<
                                        ^
/root/catkin_ov2slam/src/ov2slam/Thirdparty/Sophus/sophus/se3.hpp:89:26: error: ‘ReturnScalar’ was not declared in this scope
   using SE3Product = SE3<ReturnScalar<OtherDerived>>;
                          ^
/root/catkin_ov2slam/src/ov2slam/Thirdparty/Sophus/sophus/se3.hpp:89:39: error: template argument 1 is invalid
   using SE3Product = SE3<ReturnScalar<OtherDerived>>;
                                       ^
/root/catkin_ov2slam/src/ov2slam/Thirdparty/Sophus/sophus/se3.hpp:92:32: error: ‘ReturnScalar’ was not declared in this scope
   using PointProduct = Vector3<ReturnScalar<PointDerived>>;
                                ^
/root/catkin_ov2slam/src/ov2slam/Thirdparty/Sophus/sophus/se3.hpp:92:45: error: template argument 1 is invalid
   using PointProduct = Vector3<ReturnScalar<PointDerived>>;
                                             ^
/root/catkin_ov2slam/src/ov2slam/Thirdparty/Sophus/sophus/se3.hpp:95:43: error: ‘ReturnScalar’ was not declared in this scope
   using HomogeneousPointProduct = Vector4<ReturnScalar<HPointDerived>>;
                                           ^
/root/catkin_ov2slam/src/ov2slam/Thirdparty/Sophus/sophus/se3.hpp:95:56: error: template argument 1 is invalid
[... 280 lines cut ...]
               ^
/root/catkin_ov2slam/src/ov2slam/Thirdparty/Sophus/sophus/se3.hpp:335:15: error: ‘HomogeneousPointProduct’ does not name a type
[... cut ...] :internal::traits<T>::Scalar = double]’:
/root/catkin_ov2slam/src/ov2slam/src/camera_calibration.cpp:178:29:   required from here
/root/catkin_ov2slam/src/ov2slam/Thirdparty/Sophus/sophus/se3.hpp:210:35: error: no match for ‘operator*’ (operand types are ‘Sophus::SO3<double>’ and ‘const ScalarMultipleReturnType {aka const Eigen::CwiseUnaryOp<Eigen::internal::scalar_multiple_op<double>, const Eigen::Matrix<double, 3, 1> >}’)
     return SE3<Scalar>(invR, invR * (translation() * Scalar(-1)));

```

ferreram, 2021-12-06:
@DandyLorenz Make sure that Sophus is indeed getting built with EIgen >= 3.3 (see https://github.com/strasdat/Sophus/issues/187).

closed 2023-06-30 by aeudes