ov2slam/ov2slam #36: terminate called after throwing an instance of 'std::out_of_range'

makintuo, 2021-12-13:
Hi,

I always get the following error at some random frame on a specific dataset I'm using. I haven't modified any of the relevant functions in VisualFrontEnd. The dataset is quite large with 10,400 stereo images and resolution of 1006x597px (resized from 2013x1195px). Alternatively, if the error doesn't pop in the beginning or middle of the trajectory, the algorithm slows down considerably and freezes the computer, requiring a forced reboot. Any idea what could be the cause? 

------------------------------------------------------------

terminate called after throwing an instance of 'std::out_of_range'
  what():  _Map_base::at
Stack trace (most recent call last) in thread 46957:
#14   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
#13   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f73f06d9292, in clone
#12   Object "/lib/x86_64-linux-gnu/libpthread.so.0", at 0x7f73f0bc9608, in 
#11   Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f73f089ade3, in 
#10   Object "/home/kamijolab/catkin_ws/devel/lib/libov2slam.so", at 0x7f73f111b2c8, in SlamManager::run()
#9    Object "/home/kamijolab/catkin_ws/devel/lib/libov2slam.so", at 0x7f73f1136795, in VisualFrontEnd::visualTracking(cv::Mat&, double)
#8    Object "/home/kamijolab/catkin_ws/devel/lib/libov2slam.so", at 0x7f73f11362e4, in VisualFrontEnd::trackMono(cv::Mat&, double)
#7    Object "/home/kamijolab/catkin_ws/devel/lib/libov2slam.so", at 0x7f73f11323ae, in VisualFrontEnd::kltTracking()
#6    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f73f086537d, in std::__throw_out_of_range(char const*)
#5    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f73f086e6a8, in __cxa_throw
#4    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f73f086e3f6, in std::terminate()
#3    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f73f086e38b, in 
#2    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f73f0862910, in 
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f73f05dc858, in abort
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f73f05fd18b, in gsignal
Aborted (Signal sent by tkill() 46938 1000)

makintuo, 2021-12-14:
Hi Maxime,

the cause is indeed excessive RAM usage (ballooning from 5GB to 32GB (max) used in short period). I am using the 'accurate' yaml parameters you provided in the parameters folder, just changing the camera and dataset details (not touching the SLAM parameters except for force_realtime which is set to 0).

However I should mention that I am using the "read images from folder" fork due having all my datasets in image format. But, none of my other datasets exhibit this problem (this dataset is however the longest one both in image count and trajectory length). The RAM usage seems to increase exponentially the longer the map becomes.

I'll try to find a way to convert the images to a ROS bag and see if the problem persists (whether its due to the different fork). Meanwhile I'll attach my parameters file just in case

[winter_seq6_night.txt](https://github.com/ov2slam/ov2slam/files/7710076/winter_seq6_night.txt)
.

ferreram, 2021-12-14:
What might happen is that too many images are selected to become a keyframe at some point and the queue used in the Mapper and Loop Closer gets saturated (the images are sent in these queues for stereo matching & loop closing features extraction).  You could try to disable the loop closure option to see if it solves the problem or not.  Also, you can enable the debug option to have a closer look at what is going on when the algorithm starts slowing down.

I have not tested the fork with the option for reading images from a folder so I cannot tell if a problem is coming from it or not.  Although, I see that a fps parameter has been added when using this option.  You should ensure that a "sleep" is occurring between the reading of each stereo pair to respect the fps set in the option, otherwise the algorithm will process a new image as soon as the front-end is done with the previous one, which is likely to saturate the algorithm with the 'accurate' config.

makintuo, 2021-12-16:
The visibility in the images is very poor (only a few meters on a road with very few visible features) due to night-time, so the system indeed seems to constantly select images as keyframes and perhaps flooding the RAM. I tested with a much larger dataset (43 000 stereo pictures taken at 20hz) with daytime lightning conditions, and the system completed it most of the time with no problems (although on few occasions the error I posted above occurred). ROS visualizer also consumes quite a bit of RAM so I ran it without it. 

Even then, only with this particular (night-time) dataset I've seen excessive RAM usage and system crashing. I tried disabling the Loop Closer but it didn't make any difference. I'll try tweaking the keyframe insertion conditions to see if it helps.

Can elaborate on the "sleep" part a bit? Do you mean making the algorithm wait between reading the stereo image pairs for example using the usleep() function?

Also, regarding the colorization of keypoints in ROS visualization, what's the difference between red, blue and green points? I didn't see this mentioned in your paper.

Thanks again for taking the time.

ferreram, 2021-12-16:
About the sleep, I indeed meant to use a usleep() function or something similar to make sure that the images are truly read at 20 Hz.  

About the colorization in Rviz, blue points are not yet triangulated keypoints, red points are triangulated ones (i.e. those used for pose estimation) and the green ones correspond to "re-tracked" keypoints which are keypoints matched to an already existing 3D map point.

GopiRajuMatta, 2023-03-22:
At the starting itself, I am getting the above error. Any work around for this?