ov2slam/ov2slam #38: Can I save the map and load it for locaization?

Taeyoung96, 2022-02-25:
@ferreram   
Thanks for your great work!    
I have some question.  

Is there any function that save the map and load it for localization?   
If not, I want to create functions by referring to [TUMFTM/orbslam-map-saving-extension](https://github.com/TUMFTM/orbslam-map-saving-extension).  
 
But, I'm not familiar with your code...  
What part of the code can be modified to create the desired functions?  

Would you mind give me some tips?  

Thanks,

ferreram, 2022-02-28:
Hi,
Right now there is no feature like this in ov2slam.  The first thing to do in order to implement map loading for localization is to modify the ibow-lcd thirdparty library's code so that you can save the created vocabulary trees and then reload them on next runs.  This step is the biggest piece of work I think.  Once this is done, it should be quite easy to use ov2slam for localization from a previously built map.

Taeyoung96, 2022-02-28:
@ferreram   
Thanks for your comment! 😆😆  
Your advice was really helpful.  

Thanks,

closed 2022-02-28 by Taeyoung96