ov2slam/ov2slam #40: Segmentation fault while running (Address not mapped to object)

maazmb, 2022-03-18:
Hi,
Firstly thanks for sharing your project for others to learn from. After successfully building the code as per instructions im facing this error while running it.
My environment is:
 ubuntu 18.04,
 ros melodic 1.14.12,
 opencv 4.5.5,
 eigen 3.3.4
opengv
ceres 2.0

I have already tried building opengv, ceres and ov2slam with turning off --march native but the same error persisted. Any suggestion would be highly appreciated. Thanks..:)

the error:
`[100%] Linking CXX executable /mnt/HD1/maazshare2/slam/Catkinov2slam/devel/lib/ov2slam/ov2slam_node
/usr/bin/ld: warning: libopencv_core.so.405, needed by /maazshare2/slam/Catkinov2slam/devel/lib/libov2slam.so, may conflict with libopencv_core.so.3.2
[100%] Built target ov2slam_node
maaz@maaz:/maazshare2/slam/Catkinov2slam$ source /maazshare2/slam/Catkinov2slam/devel/setup.bash
maaz@maaz:/maazshare2/slam/Catkinov2slam$ rosrun ov2slam ov2slam_node /maazshare2/slam/Catkinov2slam/src/ov2slam/parameters_files/accurate/kitti/kitti_00-02.yaml

Launching OV²SLAM...

Loading parameters file : /maazshare2/slam/Catkinov2slam/src/ov2slam/parameters_files/accurate/kitti/kitti_00-02.yaml...

Parameters file loaded...

SLAM Parameters are being setup...
Stack trace (most recent call last):
#14   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
#13   Object "/maazshare2/slam/Catkinov2slam/devel/lib/ov2slam/ov2slam_node", at 0x55b6f64350c9, in _start
#12   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f79fd4f7bf6, in __libc_start_main
#11   Object "/maazshare2/slam/Catkinov2slam/devel/lib/ov2slam/ov2slam_node", at 0x55b6f643562a, in main
#10   Object "/maazshare2/slam/Catkinov2slam/devel/lib/libov2slam.so", at 0x7f79ff3e645c, in SlamParams::SlamParams(cv::FileStorage const&)
#9    Object "/usr/local/lib/libopencv_core.so.405", at 0x7f79f9f28e38, in cv::FileNode::operator int() const
#8    Object "/usr/local/lib/libopencv_core.so.405", at 0x7f79f9f27d2d, in cv::FileStorage::Impl::getNodePtr(unsigned long, unsigned long) const
#7    Object "/usr/local/lib/libopencv_core.so.405", at 0x7f79f9f70ae9, in cv::error(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char const*, char const*, int)
#6    Object "/usr/local/lib/libopencv_core.so.405", at 0x7f79f9f708c9, in cv::Exception::Exception(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int)
#5    Object "/usr/lib/x86_64-linux-gnu/libopencv_core.so.3.2", at 0x7f79fea14e2b, in cv::Exception::formatMessage()
#4    Object "/usr/lib/x86_64-linux-gnu/libopencv_core.so.3.2", at 0x7f79fea14c4d, in cv::format(char const*, ...)
#3    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f79fd607ff8, in __vsnprintf_chk
#2    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f79fd532db9, in _IO_vfprintf
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f79fd530b44, in 
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f79fd6644e1, in 
Segmentation fault (Address not mapped to object [0x1e])
Segmentation fault (core dumped)` 

*Note: On debugging the error comes while reading from the parameter file in slam_params.cpp:
`debug_ = static_cast<int>(fsSettings["debug"]);;`

maazmb, 2022-03-18:
the error seems to be between the ros melodic opencv-bridge(3.2) and system opencv(4.5).

ferreram, 2022-03-18:
Hi,
You cannot mix different versions of OpenCV (i.e. 3.2 and 4.5). You should enforce the use of OpenCV 3.2 by modifying the CMakeLists.txt : find_package(OpenCV REQUIRED) ==> find_package(OpenCV 3.2 REQUIRED).  See #17 and #30 for related issue.

closed 2022-03-21 by maazmb