ov2slam/ov2slam #43: Question about precision

canglangzhige, 2022-06-06:
Hi. I have a few questions:
1. Is it possible to set body_T_cam0 to I_4x4 and body_T_cam1 to T_cam0_cam1.
2. Why is it calculated that the translation part tx in T_cam0_cam1 is a positive number, we take the left camera as the origin, shouldn’t the translation transformation of the right camera relative to the left object be a negative number?
3. I run this algorithm with the dataset and it works well, even if I deliberately change body_T_cam0 or body_T_cam1, I can get the running result, the difference is that the trajectory is not accurate. But when I run the algorithm with my own camera (realsense d455 and ZED2) in hand, the trajectory drifts very badly

canglangzhige, 2022-06-06:
> Hi. I have a few questions:
>
>     1. Is it possible to set body_T_cam0 to I_4x4 and body_T_cam1 to T_cam0_cam1.
>
>     2. Why is it calculated that the translation part tx in T_cam0_cam1 is a positive [...]

To add, I have already calibrated the camera.

aeudes, 2022-06-07:
Hi,
1. yes it should not be a problem to have the cam & body at the same frame (see tartan air stereo parameter for example)
2. The camera frame in ov²slam uses the standard computer vision convention with z forward x right y down. So for the translation between camera left(cam0) and right(cam1) expressed in left(cam0), having a tx positive is expected.
3. Maybe, if the relative position between cameras are incorrect, ov²slam will not be able to correctly triangulate stereo points and relies only on monocular slam: it is less accurate and the result is at a scale factor. You could check that in the log output, if there is a problem with relative position of cameras there will be no stereo points kept after the bundle adjustment step.

Hope this helps, if not could you provided some additional information:
- do you use color or IR cameras in the d455
- what tools do you used to make the calibration

canglangzhige, 2022-06-08:
1. I just used the left and right cameras, turned off the IR cameras, it is said that the IR flare will affect the quality of the left and right camera images.
2. I used two methods for calibration: a. kalibr; b. using the ROS package "image_pipeline". Method b is more accurate
3. At the end of the run, 5 traj files will be obtained, as shown below:
[image]
4. Which trajectory should be used if comparing with public datasets (eg EuRoC). Why don't "vo_traj" and "kfs_traj" overlap when observed in rviz, shouldn't kfs be included in all image frames, so "kfs_traj" should be included in "vo_traj"?

ferreram, 2022-06-23:
Hi @canglangzhige,

The kfs_traj is computed from the true position of each keyframe while vo_traj is simply a vector of camera positions that contains the position estimated in real-time.  Hence, if a loop closure is detected, the past positions stored in vo_traj will not be updated whereas the positions in kfs_traj will be updated according to the correction applied after the loop closure detection.

For comparing to public datasets, ov2slam_kfs_traj is most of the time the right choice.

closed 2022-11-25 by ferreram

KomalBagai, 2023-06-30:
Hello, 
I am facing the similar issue with the drift and false BA and LC everywhere with the ZED camera. I noticed the problem is worst when it misses a frame or two while detecting keypoints and then the trajectory just goes crazy. I have tried:
1. switching off the loop closure
2. reducing nmaxdist
3. using fast
4. using shi tomasi

I am doing this on the image below, can you point me to correct feature detection parameter so as to get reliable features consistently?
Any help is much appreciated.
[image]

Thanks,