ov2slam/ov2slam #44: Error: R is not orthogonal with Realsense T265

novadr, 2022-10-18:
Hello, 

I'm trying to run OV2SLAM with an Intel Realsense T265, but I'm facing an issue with my yaml file. As you can see below, I set the camera model as fisheye and complete the file according to the parameters I got with the command : _rs-enumerate-devices -c_ 

I also set the extrinsic parameters according to the transformation between the left and the right fisheye cameras.

```
#--------------------------------------------------------------------------------------------
# Camera Parameters. Adjust them!
#--------------------------------------------------------------------------------------------

Camera.topic_left: /camera/fisheye1/image_raw
Camera.topic_right: /camera/fisheye2/image_raw

Camera.model_left: fisheye
Camera.model_right: fisheye

Camera.left_nwidth: 848
Camera.left_nheight: 800

Camera.right_nwidth: 848
Camera.right_nheight: 800

# Camera calibration and distortion parameters (OpenCV) 
Camera.fxl: 287.854705810547
Camera.fyl: 287.795593261719
Camera.cxl: 426.878387451172
Camera.cyl: 400.959686279297

Camera.k1l: -0.0112883998081088
Camera.k2l: 0.0493851415812969
Camera.p1l: -0.0468114204704762
Camera.p2l: 0.00931846909224987

Camera.fxr: 287.768005371094
Camera.fyr: 287.635314941406
Camera.cxr: 421.719085693359
Camera.cyr: 394.745513916016

Camera.k1r: -0.0121076302602887
Camera.k2r: 0.0504187308251858
Camera.p1r: -0.0474353283643723
Camera.p2r: 0.00939208921045065

# Camera Extrinsic parameters T_b_ci ( v_b = T_b_ci * v_ci )
body_T_cam0: !!opencv-matrix
   rows: 4
   cols: 4
   dt: d
   data:  [0.99998, -0.000684938, 0.00627555, 0.0637358278036118,
           0.000680629, 1.0    , 0.000688822, -0.000107772735645995,
          -0.00627602, -0.000684538 , 0.99998, -0.000309090450173244,
          0.0, 0.0, 0.0, 1.0]

body_T_cam1: !!opencv-matrix
   rows: 4
   cols: 4
   dt: d
   data: [1., 0., 0., 0,
          0., 1., 0., 0,
          0., 0., 1., 0,
          0., 0., 0., 1.]
```

OV2SLAM has been built successfully, and it actually runs with the different euroc yaml files. But when I try to use my yaml file it gives me an error with Sophus : 

[image]

Has somebody else encounter this error, please ? I really don't know what is the origin of this issue and how to fix that…

Thank you

aeudes, 2022-10-19:
When Sophus build the SE3 object from the matrix input, it checks if the rotation part is an orthogonal matrix and yours is not at the required precision.
You can check this by computing the residual of R*R^T - I and R^T*R -I 

One way to re-normalized the matrix is to apply SVD on it.
python code:
```
import numpy as np
U,D,V = np.linalg.svd(R)                                        

R2 = U@V.T 
```

novadr, 2022-10-20:
Hello aeudes, thank you for your answer

I am now able to locate the problem in the code. It happens in the file **slam_params.cpp** when we create the Sophus::SE3d object. 
` T_left_right_ = Sophus::SE3d(Tbc0.inverse() * Tbc1); `

I tried to "re-normalized" the matrix with the SVD method as you said, but it didn't change anything. 
When I print in terminal the rotation matrix before SVD and the matrix after the SVD it has the same values.
Here is the code I add in the file : 
```
        // ADD
        Eigen::Matrix4d result;
        result = Tbc0.inverse() * Tbc1;
        Eigen::MatrixXd normalized_rotation_matrix = Eigen::MatrixXd::Identity(3, 3);
        normalized_rotation_matrix << result.coeff(0, 0), result.coeff(0, 1), result.coeff(0, 2), result.coeff(1, 0), result.coeff(1, 1), result.coeff(1, 2), result.coeff(2, 0), result.coeff(2, 1),result.coeff(2, 2);
        Eigen::JacobiSVD<Eigen::MatrixXd> svd( normalized_rotation_matrix, Eigen::ComputeFullU | Eigen::ComputeFullV);
        Eigen::MatrixXd rot = svd.matrixU() * (svd.matrixV()).transpose();

        T_left_right_ = Sophus::SE3d(rot);
```

Maybe I misunderstood something ? What do you mean exactly by "re-normalized" the matrix ? The expected values should be close to an Identity matrix because the rotation between the two cameras is almost an identity matrix, is that right ?

Moreover when I launch the _euroc_stereo.yaml_ ,  even if the matrix is not perfectly orthogonal the algorithm runs.

[image]

I tried to round the values of the rotation matrix and the algorithm ran but it wasn't able to compute the trajectory. All of the features was blue on the image in rviz.

ferreram, 2022-10-24:
Hi @novadr ,

To solve your problem, you can simply create the extrinsic T_left_right_ matrix as follow :
```cpp
Sophus::SE3d(Sophus::SO3d::fitToSO3(Rot_matrix), translation_vector)
```
This way, Sophus will automatically fit the rotation matrix Rot_matrix so that it fits the property of SO(3).

Otherwise, as @aeudes pointed out, you can compute the SVD on the rotation matrix.  This gives you the following results :
```python
U,D,V = np.linalg.svd(R)
newR = U.dot(V)

array([[ 9.99980074e-01, -6.84938113e-04,  6.27555103e-03],
       [ 6.80628614e-04,  9.99999531e-01,  6.88822385e-04],
       [-6.27601988e-03, -6.84537340e-04,  9.99980071e-01]])
```
Where the newR is orthogonal and of determinant 1.

novadr, 2022-10-26:
Hello @ferreram,

I try to modify the code with fitToSO3 as you said. It works fine, thank you very much ! 

For thoses who get the same problem, this is what I modify in the code **slam_params.cpp**  : 

I replaced this (line 87) : 

`T_left_right_ = Sophus::SE3d(Tbc0.inverse() * Tbc1);`

By : 

```
Eigen::Matrix4d result = Tbc0.inverse() * Tbc1;
Eigen::MatrixXd rotation_matrix = Eigen::MatrixXd::Identity(3, 3);
Eigen::VectorXd translation_vector = Eigen::VectorXd::Zero(3);
rotation_matrix << result.coeff(0, 0), result.coeff(0, 1), result.coeff(0, 2), result.coeff(1, 0), result.coeff(1, 1), result.coeff(1, 2), result.coeff(2, 0), result.coeff(2, 1),result.coeff(2, 2);
translation_vector << result.coeff(0, 3), result.coeff(1, 3), result.coeff(2, 3);

T_left_right_ = Sophus::SE3d(Sophus::SO3d::fitToSO3(rotation_matrix), translation_vector);
```
I also notice that my yaml file was not correct.

For the transformation between the two camera : 
The good way is to set **body_T_cam0** as the transformation from left camera to right camera and **body_T_cam1** as an identity matrix. 

In my case : 
`body_T_cam0: !!opencv-matrix
   rows: 4
   cols: 4
   dt: d
   data:  [0.99998, 0.000680629, -0.00627602, -0.0637364163994789,
          -0.000684938 , 1.0    , -0.000684538, 0.000151216212543659,
           0.00627555, 0.000688822, 0.99998, -9.08187066670507e-05,
           0.0, 0.0, 0.0, 1.0]

body_T_cam1: !!opencv-matrix
   rows: 4
   cols: 4
   dt: d
   data: [1., 0., 0., 0,
          0., 1., 0., 0,
          0., 0., 1., 0,
          0., 0., 0., 1.]`

Now it works fine, thank you @ferreram and @aeudes !

closed 2022-10-26 by novadr