ov2slam/ov2slam #48: OpenCV Include problem

rreddy78, 2022-11-18:
Hello devs,

I am trying to build the project. I am getting this OpenCV include problem:

```
/home/sxv1kor/OV2SLAM/catkin_ws/src/ov2slam/src/feature_extractor.cpp:36:10: fatal error: opencv2/xfeatures2d.hpp: No such file or directory
   36 | #include <opencv2/xfeatures2d.hpp>
```
I have opencv-contrib installed in my system:
```
sudo apt install libopencv-contrib-dev libopencv-contrib4.2 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libopencv-contrib-dev is already the newest version (4.2.0+dfsg-5).
libopencv-contrib4.2 is already the newest version (4.2.0+dfsg-5).
0 upgraded, 0 newly installed, 0 to remove and 288 not upgraded.
```

So i explicitly included in CMakeLists.txt:

```
find_package(OpenCV REQUIRED)
include_directories( ${OpenCV_INCLUDE_DIRS} )
message(STATUS "Found OpenCV include libraries at: ${OpenCV_INCLUDE_DIRS} ")
```

And I can see that the dirs are there:
```
-- Found OpenCV: /usr (found version "4.2.0") 
-- Found OpenCV include libraries at: /usr/include/opencv4 
```

Now why is the project not including the headers ?

ferreram, 2022-11-18:
Hi @rreddy78 ,

When installing OpenCV from source code, the contrib repository must be linked to it at build time.  That is, you cannot first build OpenCV and then build the opencv-contrib modules, both have to be built together.  
I don't know how it is meant to be handled for the contrib modules with the apt install but I suspect that it cannot actually work this way since OpenCV was most likely already installed.

So, you can either try to re-install OpenCV using the source code repositories or simply set to OpenCV Contrib flag to off in OV2SLAM's CMakeLists.txt (the performances should actually be identical whether you use the contrib repository or not).

aeudes, 2022-11-18:
Hi  @rreddy78,
just a small addition, the opencv-contrib ubuntu package did not include the xfeature2d module needed by ov²slam (du to licensing problem I think).
If you really want to try it, you have to build it from source as explain by @ferreram.

closed 2022-11-25 by ferreram