ov2slam/ov2slam #49: Getting to work with CARLA Simulator

rreddy78, 2022-11-18:
Hello devs,

I am trying to get the slam work with CARLA Simulator and I have defined the parameter file (simulation.yaml in avarage folder) as follows. Other params are same as KITTI
``
Camera.topic_left: /carla/ego_vehicle/rgb_left/image
Camera.topic_right: /carla/ego_vehicle/rgb_right/image

Camera.model_left: pinhole
Camera.model_right: pinhole

Camera.left_nwidth: 1241
Camera.left_nheight: 376

Camera.right_nwidth: 1241
Camera.right_nheight: 376

# Camera calibration and distortion parameters (OpenCV) 
Camera.fxl: 854.0449816523632
Camera.fyl: 854.0449816523632
Camera.cxl: 620.5
Camera.cyl: 188.0

Camera.k1l: 0.
Camera.k2l: 0.
Camera.p1l: 0.
Camera.p2l: 0.

Camera.fxr: 854.0449816523632
Camera.fyr: 854.0449816523632
Camera.cxr: 620.5
Camera.cyr: 188.0

Camera.k1r: 0.
Camera.k2r: 0.
Camera.p1r: 0.
Camera.p2r: 0.

# Camera Extrinsic parameters T_b_ci ( v_b = T_b_ci * v_ci )
body_T_cam0: !!opencv-matrix
   rows: 4
   cols: 4
   dt: d
   data: [1., 0., 0., 0.,
          0., 1., 0., 0.,
          0., 0., 1., 0.,
           0., 0., 0., 1.]

body_T_cam1: !!opencv-matrix
   rows: 4
   cols: 4
   dt: d
   data: [1., 0., 0., 0.06,
          0., 1., 0., 0.,
          0., 0., 1., 0.,
           0., 0., 0., 1.]
``
I have camera FPS at 20Hz in Simulator. 

```
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.2.0) ../modules/video/src/lkpyramid.cpp:1391: error: (-215:Assertion failed) prevPyr[level * lvlStep1].size() == nextPyr[level * lvlStep2].size() in function 'calc'

Stack trace (most recent call last) in thread 307701:
#16   Object "", at 0xffffffffffffffff, in 
BFD: DWARF error: section .debug_info is larger than its filesize! (0x93ef57 vs 0x530ea0)
#15   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fa5bcebb132, in clone
#14   Source "/build/glibc-SzIz7B/glibc-2.31/nptl/pthread_create.c", line 477, in start_thread [0x7fa5bd3ad608]
#13   Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fa5bd081de3, in std::error_code::default_error_condition() const
#12   Source "/home/sxv1kor/OV2SLAM/catkin_ws/src/ov2slam/src/mapper.cpp", line 83, in Mapper::run() [0x7fa5bd9248a6]
         80:                 std::vector<cv::Mat> vpyr_imright;
         81:                 cv::buildOpticalFlowPyramid(imright, vpyr_imright, pslamstate_->klt_win_size_, pslamstate_->nklt_pyr_lvl_);
         82: 
      >  83:                 pmap_->stereoMatching(*pnewkf, kf.vpyr_imleft_, vpyr_imright);
         84:                 
         85:                 if( pnewkf->nb2dkps_ > 0 && pnewkf->nb_stereo_kps_ > 0 ) {
#11   Source "/home/sxv1kor/OV2SLAM/catkin_ws/src/ov2slam/src/map_manager.cpp", line 550, in MapManager::stereoMatching(Frame&, std::vector<cv::Mat, std::allocator<cv::Mat> > const&, std::vector<cv::Mat, std::allocator<cv::Mat> > const&) [0x7fa5bd915cf1]
        547:         // Good / bad kps vector
        548:         std::vector<bool> vkpstatus;
        549: 
      > 550:         ptracker_->fbKltTracking(
        551:                     vleftpyr, 
        552:                     vrightpyr, 
        553:                     pslamstate_->nklt_win_size_, 
#10   Source "/home/sxv1kor/OV2SLAM/catkin_ws/src/ov2slam/src/feature_tracker.cpp", line 66, in FeatureTracker::fbKltTracking(std::vector<cv::Mat, std::allocator<cv::Mat> > const&, std::vector<cv::Mat, std::allocator<cv::Mat> > const&, int, int, float, float, std::vector<cv::Point_<float>, std::allocator<cv::Point_<float> > >&, std::vector<cv::Point_<float>, std::allocator<cv::Point_<float> > >&, std::vector<bool, std::allocator<bool> >&) const [0x7fa5bd910db2]
         63:     vkpsidx.reserve(nbkps);
         64: 
         65:     // Tracking Forward
      >  66:     cv::calcOpticalFlowPyrLK(vprevpyr, vcurpyr, vkps, vpriorkps, 
         67:                 vstatus, verr, klt_win_size,  nbpyrlvl, klt_convg_crit_, 
         68:                 (cv::OPTFLOW_USE_INITIAL_FLOW + cv::OPTFLOW_LK_GET_MIN_EIGENVALS) 
         69:                 );
```

ferreram, 2022-11-18:
Are you sure of your images size ? Also, make sure than both the left and right images have the same size!

rreddy78, 2022-11-18:
Oh right! I got the image sizes wrong in CARLA. left image size != right image size.
Thank you. Working fine now. 

Though after loop closure the visualization goes haywire.. Unfortunately cannot upload image at at moment.

rreddy78, 2022-11-18:
Definitely there seems lot of drift and after loop closure there seems to be a problem. Will open a separate issue

closed 2022-11-18 by rreddy78