ov2slam/ov2slam #50: Rviz visualization

GopiRajuMatta, 2022-11-24:
Hello,

"Visualize OV²SLAM outputs in Rviz by loading the provided configuration file: ov2slam_visualization.rviz."

Can you please update the read me with the corresponding command, it would be helpful.

Thank you
Gopi

ferreram, 2022-11-24:
Hi @GopiRajuMatta ,

There is no need for a command here, the config file can be directly loaded from the rviz GUI:  File --> Open Config

GopiRajuMatta, 2022-11-24:
Thank you @ferreram 

I am very new to ros platform. 

Thank you
Gopi

GopiRajuMatta, 2022-11-24:
Also @ferreram 

I am planning to run this on Kitti dataset. Do I need to create a bag file to run on kitti or can I directly run on kitti image sequence? Help in this regard is much appreciated.

Thank you
Gopi

ferreram, 2022-11-24:
You need to create bag files with OV2SLAM.  However, there is a pull request that handles reading images from folders.  You can check it out but I have not tested it.

GopiRajuMatta, 2022-11-25:
Thank you so much, I will check it out...

closed 2022-11-25 by ferreram