ov2slam/ov2slam #54: question about enforcing real-time condition

jaehwan-pi, 2023-05-19:
Thank you for sharing nice work.
I have question about your paper.
How did you enforce real-time condition on your simulation?
For example, when you test orb-slam3 with real-time mode, have you simulated real-time dataset publisher such as ROS?
I could not find detailed explanation on this.

ferreram, 2023-05-20:
Hi, 
ROS was indeed used to enforce real-time.  
About these 3 files:

-  ov2slam_traj.txt : is the true real-time trajectory
- ov2slam_kfs_traj.txt : is the final poses of the keyframes, i.e. they contain the last estimation of the keyframes' pose, accounting for all optimization that can have occured (bundle adjustment and pose graph optimization).
- ov2slam_full_traj_wlc_opt.txt : is the same as above but for all the images, not only the keyframes.  In order to propagate the pose updates that occured on the keyframes, the poses of "normal" frames have been optimized with a Pose Graph like optimization.

Best

jaehwan-pi, 2023-05-22:
Thank you for your reply.
Now I understand that ROS is used for  real-time mode.
When you test ORB-SLAM3, ROS is used as well?
I'd like to know how to enforce real-time on ORB-SLAM3.

ferreram, 2023-05-22:
I have not used ORB-SLAM3 but ORB-SLAM2 to compare to OV2SLAM in the paper.  However, it seems that the ROS examples in ORB-SLAM3 are directly taken from the ORB-SLAM2 implementation so that should work.  In any case, to ensure real-time constraint make sure to set the queue parameters to 1 in the subscribe function (which is already the way it was implemented in ORB-SLAM2).

jaehwan-pi, 2023-05-23:
Thank you very much.
Just one more question!
You have implemented iBoW for loop detection.
Have you tried to use conventional BoW, which is implemented in ORB-SLAM?
I'd like to know  performances difference when using iBoW and conventional BoW.

closed 2023-05-24 by jaehwan-pi