ov2slam/ov2slam #55: "When I run this program package, it freezes."

zhifan1111, 2023-05-20:
Optimizer Object is created!

 Estimator Object is created!

 Optimizer Object is created!

 LoopCloser won't be used!

Mapper Object is created!

Mapper is ready to process Keyframes!

Sensors Grabber is created...

OV²SLAM is ready to process incoming images!

 Estimator is ready to process Keyframes!

Starting the measurements reader thread!

"There's no continuation from here."

ferreram, 2023-05-20:
Make sure to subscribe to the correct image topics in the yaml parameters file.  Your issue is that ov2slam is not receiving any image.

zhifan1111, 2023-05-21:
> Make sure to subscribe to the correct image topics in the yaml parameters file. Your issue is that ov2slam is not receiving any image.

thank you,So LoopCloser won't be used! Is it normal not to enable it?

ferreram, 2023-05-21:
In order to use the loop closer you need to set its related parameters to 1 in the yaml configuration file.  Also, you need the iBoW-LCD library to be built on your system.  You can use the build_thirdparty.sh script to do make sure the library is built.

closed 2023-06-30 by aeudes