ov2slam/ov2slam #57: differences pose data

zhifan1111, 2023-05-24:
ov2slam_full_traj_wlc_opt.txt; ov2slam_traj.txt; ov2slam_traj_kitti.txt; ov2slam_full_traj_wlc.txt.  Could you please tell me the specific differences among the above mentioned pose data?

zhifan1111, 2023-05-24:
> Please see #54

"Thank you, but I didn't download the KITTI dataset. Why was the traj_kitti.txt file generated? Also, what is the difference between ”full_traj_wlc“ and “full_traj_wlc_opt”

ferreram, 2023-05-24:
The traj_kitti.txt file is generated to give you a result in the format expected for the kitti dataset, that is one pose per image.
The full_traj_wlc.txt file contains the pose of every processed image while accounting for the potentially closed loop.
The full_traj_wlc_opt.txt file is the same as above but with a pose graph optimization applied to all the frames that are not keyframes.

zhifan1111, 2023-05-25:
> 

Thank you very much for your answer

closed 2023-06-30 by aeudes