ov2slam/ov2slam #59: Ov²slam tracking issue on Zed camera

aeudes, 2023-06-30:
Hello, 
I am facing the similar issue with the drift and false BA and LC everywhere with the ZED camera. I noticed the problem is worst when it misses some frame while detecting keypoints and then the trajectory just goes crazy. I have tried:
1. switching off the loop closure
2. reducing nmaxdist
3. using fast
4. using shi tomasi

I am doing this on the image below, can anyone point me to correct feature detection parameter so as to get reliable features consistently? @aeudes @ferreram 
Any help is much appreciated.
[image]

Thanks,

_Originally posted by @KomalBagai in https://github.com/ov2slam/ov2slam/issues/43#issuecomment-1614139664_

aeudes, 2023-06-30:
Hello,
this should be a new issue as it is totally unrelated to previous one.
It is difficult to say without a proper sequence and more details ov²slam behavior.

Never the less some points from the image and hardware info you gives:
- did you check the camera calibration ?
- ov²slam should not misses frames, could you elaborate on this ? when did it happen ? what computer did you use ? 
- your robot take a large part of the bottom of the image, you should consider dropping points tracked in this area.
Major concern :
- Up to my knowledge ZED camera are rolling shutter camera, ov²slam is not designed to work with this type of camera !
The rolling shutter effect will mess the pose computation and tracking.

KomalBagai, 2024-01-19:
I just saw this comment has been turned into an issue. Thank you for your reply. This issue can be closed now. +1

closed 2024-01-19 by ferreram