ov2slam/ov2slam #62: Don't know how to run it

TaozheLi, 2023-10-23:
How to run it ?
I run this ov2slam as following sequences after compiled:
1. new terminal:
roscore
2. new terminal:
rosrun ov2slam ov2slam_node specifically_yaml_file(And I can see it tells me starting measurements reader thread)
3. new terminal:
rosbag play kitti_data_odometry_color_sequence_00.bag
4. new terminal:
rviz -d ov2slam_visualization.rviz
But I can't see any images, and rosnode seems like not read image, it still stuck in xxxxx reader thread

aeudes, 2023-10-23:
Process seems legit...
Could you check that the camera topic in our bag and in the parameter file match and are in same namespace :
```
rosbag info kitti_data_odometry_color_sequence_00.bag
cat specifically_yaml_file | grep topic
```

Another point, when replaying bag it is usually recommended to use simulated time.
you should run ``` rosparam set /use_sim_time true ``` before starting ov2slam.
and start rosbag with simulated clock:
```
rosbag play --clock kitti_data_odometry_color_sequence_00.bag
```

TaozheLi, 2023-10-23:
Problems happened in
file feature_extractor.cpp line 258
```
pbrief_->compute(im, vkps, descs);
```
It can't compute descriptors, I try to printf ```descs```, it's ```[]```. However, detected feature which saved in ```vkps``` is not null, evidence is that I plotted key-points which saved in ```vkps``` on input_image.

aeudes, 2023-10-24:
Hum.
Could you clarify the following points:
- what is your ubuntu, ros, opencv version
- did you compile opencv from source or is it the system install
- did you build ov2slam with the OPENCV_CONTRIB flag

Most of the opencv error comes from the opencv version in ros that did not match the opencv version used to build ov2slam.
It's very strange to have keypoint but be unable to extract at least one descriptor, is the image ```im``` is correct in the call ? if you build with OPENCV_CONTRIB  you can try to deactive it.

TaozheLi, 2023-11-01:
Thanks. I've been busy last few days. I followed your advice and reinstall opencv 3.4 version, it works

closed 2023-11-01 by TaozheLi